tesseract-robotics/trajopt #16: TrajOpt crashes when using continuous collision detection

arocchi, 2018-10-03:
latest `trajopt-ros` with latest `tesseract` causes segfault when using continuous collision detection for the collision constraint.

Levi-Armstrong, 2018-10-03:
Is this using #27

arocchi, 2018-10-03:
yes, this is with ros-industrial-consortium/tesseract@38ae194b2cb60758f5c3ffe4217874109ee40511 and trajopt 5345adb8fc1cf61455009592cb68e40456c09d1d
An example: I set `"continuous" : true,` in `/trajopt_examples/config/glass_up_right_plan.json` and did `roslaunch trajopt_examples glass_up_right_plan.launch` and the output was: 
```
process[trajopt_examples_glass_up_right_plan-2]: started with pid [814]
bpmpd_caller: /home/alessio/test_ws/src/trajopt_ros/trajopt_sco/include/trajopt_sco/bpmpd_io.hpp:37: void bpmpd_io::ser(int, T&, bpmpd_io::SerMode) [with T = char]: Assertion `n == sizeof(T)' failed.
Aborted (core dumped)
```

arocchi, 2018-10-04:
Unsurprisingly, this (very) weird bug also happens in code that use continuous collision checking just for the sake of validating the result of the plan.
For instance, in `trajopt_planner.cpp` at line `106`, I call:
```
ContinuousContactManagerBasePtr manager = prob->GetEnv()->getContinuousContactManager();
```

and obtain sometimes this segmentation fault:
```
0x00007fffb45c3c40 in btCollisionObject::getBroadphaseHandle (this=0x0)
    at /home/alessio/ros_ws/src/third_party/tesseract/tesseract_ext/bullet3_ros/bullet3/src/BulletCollision/CollisionDispatch/btCollisionObject.h:391
391			return m_broadphaseHandle;
(gdb) bt
#0  0x00007fffb45c3c40 in btCollisionObject::getBroadphaseHandle (this=0x0)
    at /home/alessio/ros_ws/src/third_party/tesseract/tesseract_ext/bullet3_ros/bullet3/src/BulletCollision/CollisionDispatch/btCollisionObject.h:391
#1  0x00007fffb45c020b in tesseract::tesseract_bullet::BulletCastBVHManager::setContactRequest (this=0x19960d50, req=...)
    at /home/alessio/ros_ws/src/third_party/tesseract/tesseract_collision/src/bullet/bullet_cast_managers.cpp:785
#2  0x00007fffb45beae9 in tesseract::tesseract_bullet::BulletCastBVHManager::clone (this=0x1afe6980)
    at /home/alessio/ros_ws/src/third_party/tesseract/tesseract_collision/src/bullet/bullet_cast_managers.cpp:523
#3  0x00007fffb641afb7 in tesseract::tesseract_ros::KDLEnv::getContinuousContactManager (this=0x1a2cb7e0)
    at /home/alessio/ros_ws/src/third_party/tesseract/tesseract_ros/include/tesseract_ros/kdl/kdl_env.h:152
#4  0x00007fffb72c9f15 in mpb::TrajOptPlanner::plan (this=0x1a532770, initial_joint_positions=std::unordered_map with 6 elements = {...}, 
    goal_joint_positions=std::unordered_map with 6 elements = {...})
    at /home/alessio/ros_ws/src/real_world/robotics/planning/motion_planning/src/trajopt_planner/trajopt_planner.cpp:106
```

of course this will eventually happens also on `trajopt_examples`, if you run them multiple times:
```
#0  0x00007fffe4864c40 in btCollisionObject::getBroadphaseHandle (this=0x0) at /home/alessio/test_ws/src/tesseract/tesseract_ext/bullet3_ros/bullet3/src/BulletCollision/CollisionDispatch/btCollisionObject.h:391
#1  0x00007fffe486120b in tesseract::tesseract_bullet::BulletCastBVHManager::setContactRequest (this=0xc75310, req=...) at /home/alessio/test_ws/src/tesseract/tesseract_collision/src/bullet/bullet_cast_managers.cpp:785
#2  0x00007fffe485fae9 in tesseract::tesseract_bullet::BulletCastBVHManager::clone (this=0xbcccb0) at /home/alessio/test_ws/src/tesseract/tesseract_collision/src/bullet/bullet_cast_managers.cpp:523
#3  0x00007ffff6e18fd7 in tesseract::tesseract_ros::KDLEnv::getContinuousContactManager (this=0x88b5c0) at /home/alessio/test_ws/src/tesseract/tesseract_ros/include/tesseract_ros/kdl/kdl_env.h:152
#4  0x0000000000549e61 in main (argc=1, argv=0x7fffffffc4e8) at /home/alessio/test_ws/src/trajopt_ros/trajopt_examples/src/glass_up_right_plan.cpp:231
```

I copy-paste lines `230-236`: this is a check which is done before the optimizer is run to see if the seed trajectory is in collision:
```
  std::vector<tesseract::ContactResultMap> collisions;
  ContinuousContactManagerBasePtr manager = prob->GetEnv()->getContinuousContactManager();

  bool found = tesseract::continuousCollisionCheckTrajectory(
      *manager, *prob->GetEnv(), *prob->GetKin(), prob->GetInitTraj(), collisions);

  ROS_INFO((found) ? ("Initial trajectory is in collision") : ("Initial trajectory is collision free"));
```

The very strange thing is that this seems to happen only some times, so it might look like a race condition. What do you think?

Levi-Armstrong, 2018-10-12:
I am currently working on a fixing the trajopt_examples.

Levi-Armstrong, 2018-10-19:
@arocchi The continuous collision checking should be resolved by the latest PR merger into Tesseract https://github.com/ros-industrial-consortium/tesseract/pull/39

arocchi, 2018-10-20:
@Levi-Armstrong thanks for the fix, I will check it out ASAP. Regarding continuous collisions, they still only work with robot-environment collision?

Levi-Armstrong, 2018-10-22:
Also, I restructured the API this weekend to remove concepts that were related to MoveIt that did not make sense in the API.

Levi-Armstrong, 2018-10-27:
@arocchi Does it still crash for your application?

arocchi, 2018-11-02:
continuous collision works fine again. Thanks @Levi-Armstrong

closed 2018-11-02 by arocchi