tesseract-robotics/trajopt #23: iiwa_stack license missing

arocchi, 2018-10-22:
The iiwa model used in the examples is probably derived by https://github.com/IFL-CAMP/iiwa_stack
Redestribution of the model binary needs the [original license](https://github.com/IFL-CAMP/iiwa_stack/blob/master/LICENSE) file to be included in the `trajopt-ros` distribution

gavanderhoorn, 2018-10-22:
I'm almost 100% sure it's actually [this macro xacro](https://github.com/ros-industrial/kuka_experimental/blob/0f03b98238ade2389770cb148aca724e4fd1580c/kuka_lbr_iiwa_support/urdf/lbr_iiwa_14_r820_macro.xacro) from the `ros-industrial/kuka_experimental` package.

Including the license would still be nice, but then it should be [apache 2.0](https://github.com/ros-industrial/kuka_experimental/blob/0f03b98238ade2389770cb148aca724e4fd1580c/LICENSE).

@Levi-Armstrong is also the [author](https://github.com/ros-industrial/kuka_experimental/blob/0f03b98238ade2389770cb148aca724e4fd1580c/kuka_lbr_iiwa_support/package.xml#L26) of the package these files came from.

arocchi, 2018-10-22:
`trajopt-examples` states in `package.xml` that the license is apache 2.0, should it also be in a LICENSE file?

gavanderhoorn, 2018-10-23:
That could be done, but seeing as the package manifest already states the package is licensed under Apache 2 that should cover it.

I'm sure @Levi-Armstrong will accept a PR adding the `LICENSE` file to `trajopt_examples` if you're willing to submit one.

Levi-Armstrong, 2018-10-27:
The kuka iiwa is a copy from the `ros-industrial/kuka_experimental`.

arocchi, 2018-10-29:
I guess it is up to @Levi-Armstrong, and as previously stated both files are under the same license. What I find very professional is doing similarly to what I have seen done in Drake, e.g. https://github.com/RussTedrake/drake/blob/e49bd826240a49d6390ce441faa7b212a71c705c/manipulation/models/iiwa_description/LICENSE.TXT where the license file states where the file comes from :) I think it would be of value since people taking a look at the source code would know where the urdf comes from, and eventually which version of the file they are using :+1:

mpowelson, 2019-09-20:
I'm going to close this since these files were removed from TrajOpt and the equivalent ones in tesseract_ros_utils have a license associated with them.

closed 2019-09-20 by mpowelson