tesseract-robotics/trajopt #27: Basic cartesian example causes Eigen unaligned array assertion failure

mpowelson, 2018-10-29:
Fresh copy of the current kinetic-devel tesseract/trajopt gives me this error when I run the basic cartesian example. I've seen it for a while, but I thought it was only in the time param branch.

```
trajopt_examples_basic_cartesian_plan: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[trajopt_examples_basic_cartesian_plan-3] process has died [pid 860, exit code -6, cmd /home/mpowelson/workspaces/temp/devel/lib/trajopt_examples/trajopt_examples_basic_cartesian_plan __name:=trajopt_examples_basic_cartesian_plan __log:=/home/mpowelson/.ros/log/44dd890a-dbc5-11e8-aef7-8c1645997b24/trajopt_examples_basic_cartesian_plan-3.log].
```

gdb backtrace
```
#0  0x00007ffff3a66428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff3a6802a in __GI_abort () at abort.c:89
#2  0x00007ffff3a5ebd7 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x5cb7c8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., 
    file=file@entry=0x5cb790 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x5d4380 <Eigen::internal::plain_array<double, 4, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]")
    at assert.c:92
#3  0x00007ffff3a5ec82 in __GI___assert_fail (
    assertion=0x5cb7c8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., 
    file=0x5cb790 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=128, 
    function=0x5d4380 <Eigen::internal::plain_array<double, 4, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:101
#4  0x000000000058946d in Eigen::internal::plain_array<double, 4, 0, 32>::plain_array() ()
#5  0x000000000057b07c in Eigen::DenseStorage<double, 4, 4, 1, 0>::DenseStorage() ()
#6  0x000000000056afc4 in Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >::PlainObjectBase() ()
#7  0x000000000055ca90 in Eigen::Matrix<double, 4, 1, 0, 4, 1>::Matrix(double const&, double const&, double const&, double const&) ()
#8  0x00007ffff74f0281 in Eigen::Quaternion<double, 0>::Quaternion(double const&, double const&, double const&, double const&) ()
   from /home/mpowelson/workspaces/temp/devel/.private/trajopt/lib/libtrajopt.so
#9  0x00007ffff67c1f7b in tesseract::tesseract_ros::urdfPose2Eigen(urdf::Pose const&) ()
   from /home/mpowelson/workspaces/temp/devel/.private/tesseract_ros/lib/libtesseract_ros_kdl.so
#10 0x00007ffff67bde31 in tesseract::tesseract_ros::KDLEnv::loadDiscreteContactManagerPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/mpowelson/workspaces/temp/devel/.private/tesseract_ros/lib/libtesseract_ros_kdl.so
#11 0x00007ffff67b9ae8 in tesseract::tesseract_ros::KDLEnv::init(boost::shared_ptr<urdf::ModelInterface const>, boost::shared_ptr<srdf::Model const>) () from /home/mpowelson/workspaces/temp/devel/.private/tesseract_ros/lib/libtesseract_ros_kdl.so
#12 0x000000000054cd7e in main ()

```

Levi-Armstrong, 2018-11-03:
@mpowelson Is this still an issue?

mpowelson, 2018-11-03:
I believe so.

arocchi, 2018-11-09:
definitely [there is a memory alignment issue](https://github.com/ethz-asl/eigen_catkin/wiki/Eigen-Memory-Issues) in tesseract/trajopt. Recently I had very (un)funny encounters with this issues: [as the document states](https://github.com/ethz-asl/eigen_catkin/wiki/Eigen-Memory-Issues#there-are-issues-in-non-eigen-elements--the-program-appears-to-defy-the-core-principles-of-the-c-language), sometimes...
> there are issues in non-Eigen elements / the program appears to defy the core principles of the C++ language

...which is even more scary as things fail silently in unrelated pieces of code which make use of unrelated funcionalities (e.g., `std::vector`)

mpowelson, 2018-11-12:
I looked at this some more this morning. While they are probably needed, as far as I can tell #42 does not fix this problem. I'm getting errors on the creation of the quaternions/isometry in urdfPose2Eigen. This strikes me as pretty odd. Perhaps the problem has to do with the way we are defining shared pointers of these classes which contain Eigen members.

I found this thread which seems to be discussing a similar issue. https://github.com/RobotLocomotion/drake/issues/1854
> One thing I don't think is very clear on Eigen's official page is the fact that an aligned allocator is needed not only if a class has a fixed-size-vectorable Eigen member variable, but if the transitive closure if all member variables within the class' member variables includes a fixed-size-vectorable Eigen type. Thus, if class Foo contains a member variable of type Bar, which contains a member variable of type Eigen::Vector2d, Foo needs the EIGEN_MAKE_ALIGNED_OPERATOR_NEW macro too, not just Bar.
> 
> Furthermore, Foo must never be instantiated using make_shared and instead must use allocate_shared.

We have `typedef std::shared_ptr<KDLEnv> KDLEnvPtr` as well as, I'm sure, several other cases of similar things.

arocchi, 2018-11-12:
@mpowelson I don't think the typedef is a problem. There might be more alignment issues in tesseract, or it could be that there are more classes using fixed size eigen types in trajopt that were not properly taken care of, or it might be that there are std containers using fixed eigen types somewhere else in the code (and as also exemplified in the drake thread, you should use a special allocator in those cases)

mpowelson, 2018-11-16:
For reference, #42 and #43  along with associated Tesseract PR did not fix this issue. Basic cartesian still crashes along with  cast_cost_octomap_unit.

arocchi, 2018-11-16:
@mpowelson, @Levi-Armstrong if we coordinate a bit on this I can try and give support on the issue. Is there a comparable effort in tesseract_* similar to #42 and #43 where I can start looking from?
Also can you give some details on how to replicate this issue? It never happened to me to have such a crash. Do you run the examples in a loop or does it happen 100% of the time? I get these issues are really hard to replicate, just wanted to make sure.
What OS are you using, which version of Eigen and which compiler, would be all good information to have.
I am currently on Ubuntu 16.04, gcc 5.4.0 and Eigen 3.2.92

mpowelson, 2018-11-19:
@arocchi It happens 100% of the time. I am running Ubuntu 16.04, gcc 5.4.0 , and I have tried Eigen 3.2.92 and 3.3.5

arocchi, 2018-11-20:
@Levi-Armstrong I noticed recently https://github.com/ros-industrial-consortium/tesseract/commit/b706bf86f119a17940db601234f9d12c50beb814 that you removed one `AlignedVector` for a regular vector. Was that intentional? Why keep `AlignedVector` for `Vector4d` and not for `Vector3d`?

Levi-Armstrong, 2018-11-20:
It is only required for fixed size types where the size is a multiple of 16 bytes.

Levi-Armstrong, 2018-11-20:
This was pointed out to me by the FCL maintainers.

arocchi, 2018-11-20:
but certainly it wouldn't hurt, and maybe it could be a problem with other architectures (32 vs 64 bits) and/or compiler optimizations? My impression is that since you went to the trouble of defining all AlignedVectors it shouldn't be a problem to have one extra, if it doesn't hurt 64bits -march and it could benefit other architectures and compiler optimization flags.

mpowelson, 2019-09-26:
Closing this as this was fixed in #133. If this pops up for anyone in the future comment or reopen this issue, as we may just need to add some more tags to the cmake

closed 2019-09-26 by mpowelson