tesseract-robotics/trajopt #36: HACD dependency

ana-GT, 2018-11-03:
Hi, I tried compiling trajopt_ros and I got a compiling error:   "Project 'trajopt_tools' tried to find library 'HACD'.  The library is neither a target nor built/installed properly". I see that in Synaptic Manager there is a libhacd2.83.6. Would that be the required library or is there any other library / repository that I should install beforehand? Thanks!

Levi-Armstrong, 2018-11-03:
The library should get build in the trajopt_ext package.

ana-GT, 2018-11-03:
@Levi-Armstrong  Thank you. I tried now installing in my laptop and it is building fine. One final question: Seems Gurobi is required, is this correct? (I see that it is required to build trajopt_sco, which in turn is required to build trajopt). I wasn't planning on purchasing a license yet (I was just planning to initially test trajopt performance)

ana-GT, 2018-11-03:
Mm. This is strange. I noticed that my workspace builds fine if I have both tesseract and trajopt_ros on it. If I only have trajopt_ros, I see the HACD compilation error.

ana-GT, 2018-11-03:
Okay, I see what is going on. It seems trajopt depends on tesseract_ros, which in turns depends on tesseract_msgs, tesseract_core and tesseract_collision, which I wasn't intending to install. I had erroneously thought that tesseract depended on trajopt (or rather, used trajopt as an option for planning), not the other way around.

arocchi, 2018-11-05:
@ana-GT I have a tesseract fork laying around which does not use fcl. I did exactly for the reason that you mention: if you are using in your codebase an older fcl version, it gets messy to use both at the same time. @Levi-Armstrong what do you think is the best course of action here? Personally, since the main branch of trajopt-ros and tesseract is named kinetic-devel, it makes sense that it should be compatible with it, so installing a different version of fcl locally might be tricky. Maybe the `no-fcl` branch should be kinetic-devel and the fcl code could go into a devel branch. The new fcl won't go into melodic either, most probably it will have to wait for a yet-unknown next ros version, so we don't know how to name the branch ATM, if not a generic devel or fcl-unstable.

Levi-Armstrong, 2018-11-05:
A little history on why the upgrade. The current binary version of FCL available is broken when performing convex collision checking and they have been doing a lot of work making fixes. 

I am on board with having FCL not build as the default and having it require a build flag to enable the build. Will this work?

arocchi, 2018-11-05:
@Levi-Armstrong I think it is a bit more complicated than that, since there are package dependencies that are on `package.xml` which is, AFAIK not scriptable - meaning that package dependencies for `trajopt_collision` will ask to build the new `fcl`.
Regarding the newest fcl, yes indeed the old one is bogus, but we have bullet ATM which should work till we switch to the newest fcl (and given TRI involvement, looks like fcl will be better supported in the future, too)

gavanderhoorn, 2018-11-05:
The only stated dependency on `fcl` I can find in a `package.xml` is in `tesseract_collision` ([here](https://github.com/ros-industrial-consortium/tesseract/blob/4410348ad1c8a74db63c3ce8e3f7e81bcf76e09c/tesseract_collision/package.xml#L23)) and that is not even a correct key afaict (should be `libfcl-dev`).

@Levi-Armstrong: is `fcl_ros` a package somewhere?

As long as the `fcl` dependency is somehow fulfilled (a fake package, a place holder, a virtual package) `rosdep` will be happy and everything should be ok.

The `fcl` plugin could even be conditionally built, only if `libfcl-dev` is actually installed -- in that case we wouldn't need to encode the `build_depend` on `libfcl-dev` in the manifest at all.

arocchi, 2018-11-05:
FYI, here is what I did https://github.com/arocchi/tesseract/tree/only_bullet
Deleting the submodules is not needed, I did it to make cloning slightly faster.

ana-GT, 2018-11-05:
After discussing with the rest of our team, we decided by now to test using fcl 0.6 (we only had to add a >=0.6 requirement in CMake for our collision detection stuff to use the fcl-ros version used by trajopt).  We have been actually talking internally about migrating our planning stuff to fcl 0.6 for a while, since we are intending to work on adding a margin feature to fcl and it makes more sense to branch off the latest FCL version rather than 0.5.  The only disadvantage is that we wouldn't be using the default fcl all other ROS packages use, but well. 

@arocchi  Thanks for the pointers!

Levi-Armstrong, 2018-11-06:
@gavanderhoorn `@Levi-Armstrong: is fcl_ros a package somewhere?`

In the `tesseract_ext` package there folder called `fcl_ros` which I clone a fcl and has a package.xml to build fcl as a pure cmake project. In the package.xml I give it the package name `fcl_ros`. Although I am not sure if this is used or even valid externally. Should I change it to `libfcl-dev`?

ana-GT, 2018-11-06:
I think the fcl_ros package is not be valid externally. I am able to use the fcl library built in fcl_ros, though, by using pkgConfig and setting the PKG_CONFIG_PATH to the workspace/devel/lib/pkgconfig path (where  fcl_ros stores the fcl.pc file). Probably there are better ways to set this up.