tesseract-robotics/trajopt #39: Circular dependencies

jrgnicho, 2018-11-07:
I've noticed that the `dependencies.rosinstall` file points to the  tesseract repository while the `dependencies.rosinstall`  in the [tesseract repository](https://github.com/ros-industrial-consortium/tesseract/blob/kinetic-devel/dependencies.rosinstall) points to trajopt_ros.  I'm not familiar enough with either one of these repositories but in general only one repo should depend on the other.

ana-GT, 2018-12-18:
Agreed. The circular dependency makes it hard to test trajopt as a standalone in planning frameworks other than tesseract, as in my current test case.

jrgnicho, 2018-12-18:
It seems like trajopt implements an interface in tesseract and that's why it needs to depend on it.  I think the right approach would be to make trajopt independent of tesseract and in tesseract create a tesseract_trajopt package that glues the two together.

Levi-Armstrong, 2018-12-21:
I believe this is not an issue with trajopt/tesseract but is a limitation of rosinstall. Each package is to be treated as an isolated project and are released as such. In this case there is not a circular build dependency, but there is a rosinstall circular dependency because it operates at the repository level and not a package level. I am not sure how to solve this issue but travis-ci does not seem to have an issue so I wonder what it is doing different.

Levi-Armstrong, 2018-12-21:
I just tested this and everything worked as expected. I cloned trajopt_ros into a clean workspace then ran rosinstall on it's .rosinstall file and everything worked. How may I recreate the issue?

jrgnicho, 2018-12-21:
@Levi-Armstrong  yes I agree with you on that for the most part it is a rosinstall issue, however I think that it would be nice to be able to use trajopt independent of any framework.  I'm not familiar with the code here so untangling tesseract may be a rather involved task.
My setup up consisted of a script that would clone both tesseract and trajopt to the workspace and then calls `wstool` on the entire workspace.  Since tesseract's `.rosinstall` file list **trajopt_ros** as **trajopt** it just clones it too even though a **trajopt_ros** repository already exists in the workspace.  Then when my script runs `catkin build` it find duplicate trajopt packages and so the build fails inmediately.  For now my script just deletes the duplicate trajopt repo from the workspace.

closed 2020-12-12 by Levi-Armstrong