tesseract-robotics/trajopt #45: Implement QuadExpr Constraints

mpowelson, 2018-11-27:
As noted in [modelling.hpp](https://github.com/ros-industrial-consortium/trajopt_ros/blob/kinetic-devel/trajopt_sco/include/trajopt_sco/modeling.hpp#L59), only affine constraints are currently implemented. I don't know what the implications of this are, but intuitively it would be nice to (for instance) but a constraint on velocity^2.

arocchi, 2018-12-07:
I remember pointing this out already, but please keep in mind that among the supported solvers only `GUROBI` allows for quadratic constraints, we would need to implement a `QCP`/`QCQP`/`SOCP` solver, I think especially the latter would be useful to have in the long-run.

mpowelson, 2018-12-07:
That's good to know. That probably explains why it hadn't been done yet. As far as the other solvers, that's outside of my area. This just came up when we were refactoring the joint constraints. I imagine Levi would be open to suggestions.