tesseract-robotics/trajopt #56: terminate called after throwing an instance of 'class_loader::LibraryUnloadException'   what():  Attempt to unload library that class_loader is unaware of.

lakshmip001, 2018-12-26:
I always come across this error when trying to launch trajopt_examples launch files. Kindly please help me why I face this problem and how I can slove this

mpowelson, 2018-12-26:
I also have been getting this error but only when shutting them down. Hence I haven't worried about it too much. It looks like this was a known issue with moveit, but I don't know much about that. https://answers.ros.org/question/218945/a-trouble-with-the-training-unit-36-from-ros-i/

lakshmip001, 2018-12-27:
Actually I am trying to visualize the basic_cartesian_plan.cpp but always it results in this error and unable to see the plan or the poses. Can you please help me how to visualize the plan and poses?

mpowelson, 2018-12-27:
I see. If you open rviz and add a Tesseract Trajectory message you should be able to see the trajectory being optimized (as long as plotting is turned on - which it should be by default). Then if you also add a basic marker and see the target points, collisions, etc. I think it is optimizing and the error is happening at shutdown. It just appears to not be doing anything because you aren't visualizing it.

lakshmip001, 2018-12-28:
I turned on the plot to visualize the plotting of the optimized I am just able to see interactive markers but no trajectory is visualized. Kindly please check the following images for reference. 
 first I am launching basic_cartesian_plan.launch
second I am running rviz and added tesseract trajectory  & pose array.
third I am running the node (trajopt_examples_basic_cartesian_plan) with param plotting:=true.
[image]
[image]
[image]