tesseract-robotics/trajopt #57: Gurobi, osqp & qpoases reg

lakshmip001, 2018-12-27:
Gurobi, OSQP and qpOASES are this installations compulsory for trajopt_ros to work. Because I didnt install any of them but optimization is done when I run the program but crashes. Kindly please help me with the instructions to install Gurobi, osqp & qpoases with ros.

mpowelson, 2018-12-27:
I can't help with install instructions as I haven't done it yet, but these are not required. BPMPD is included in the repo, so that is what you are seeing being used now.

lakshmip001, 2018-12-28:
Thanks . I thought my error was because of the optimizer so I thought of try changing it .

mpowelson, 2019-01-02:
If you're still interested in trying other optimizers, I just ran it with OSQP for the first time. The install instructions on their website worked fine. From [Build From Sources](https://osqp.org/docs/get_started/sources.html#build-from-sources):
```
git clone --recursive https://github.com/oxfordcontrol/osqp
cd osqp
mkdir build
cd build
cmake -G "Unix Makefiles" ..
cmake --build .
sudo cmake --build . --target install
```
Remove via
```
cmake --build . --target uninstall
```

lakshmip001, 2019-01-03:
Thanks.I will try doing it.

mpowelson, 2019-01-03:
I may have had something else wrong, but the catkinized qpOASES works for me. 

@lakshmip001 Here are instructions to set up qpOASES.

1. Clone [this repo](https://github.com/arocchi/qpOASES/tree/catkinized) into your workspace. Make sure you grab his catkinized branch
2. Add export TRAJOPT_CONVEX_SOLVER=QPOASES to your .bashrc file
3. clean and rebuild workspace