tesseract-robotics/trajopt #58: Optimized Trajectory output

lakshmip001, 2019-01-02:
Hi, 
I am able to sent trajectory using .json file to optimize but in the optimized results I am unable to find optimized trajectory. Kindly please guide me if I am wrong and please help me to visualize the trajectory. So That I can give it to the robot for visualization.

mpowelson, 2019-01-02:
This depends on how you set up the problem. If you are using the method in the examples, then do something like this. This will rearrange the optimization vector (now the solution if you have optimized) into an Eigen matrix with the rows being time steps and the columns being associated with the vars (in this case the joint values)
```
sco::BasicTrustRegionSQP opt(prob);
// Add callbacks, initialize, etc
...
// Get trajectory
TrajArray output = getTraj(opt.x(), prob->GetVars());
```

If you are using the tesseract planner interface then the trajectory is stored in the PlannerResponse you gave the planner like so
```
tesseract::tesseract_planning::TrajOptPlannerConfig config(prob);
tesseract::tesseract_planning::PlannerResponse planning_response;
// Solve problem
planner.solve(planning_response_place, config_place);
// plot the trajectory in Rviz
plotter.plotTrajectory(env->getJointNames(), planning_response.trajectory.leftCols(env->getJointNames().size()));
//View trajectory
std::cout << planning_response_place.trajectory << '\n';
```

lakshmip001, 2019-01-03:
Thanks!! I used the eigen matrix method given in trajopt examples . When I use this method tesseract trajectory in the rviz doesn't get the trajectory and need to pass it to plot it or move the arm. Orelse I need to program using everything through tesseract then I can get my trajectory in tesseracttrajectory in rviz. Does this one use moveit(move_group) for moving the robot arm? Kindly please guide if I am wrong through an example for the second method so that I can use tesseracttrajectory plugin on rviz.

lakshmip001, 2019-01-03:
Actually I tried including tesseract_planning/header but while I try to add it says header file isnt found while I am able to use other header files in tesseract_ros. I will try the example provided in the above. I tried eigen method obtained the trajectory and stored in the trajectory_msgs. And published the trajectory to the follow_joint_trajectory. I was able to see the robot moving, but I wasn't able to use tesseract trajectory and doubt whether this package uses moveit package or is it totally different? Can I adjust the parameters in .json file to move in a straight line from point to point with a trapezoidal velocity profile or constant velocity?

mpowelson, 2019-01-03:
This does not use moveit. You can add a velocity cost or constraint to specific values for specific time steps. You can currently set both joint and cartesian velocities without time parameterization, and it will use a unit time step.

Time parameterization is in the works. See #51 . Joint level cost/cnt should work if you want to pull that branch.

lakshmip001, 2019-01-03:
Thanks I will apply and see.Please help me in addition of tesseract_planning header into my code.

lakshmip001, 2019-01-04:
Thanks got it. Can you provide example code for giving 3 poses with velocity constraints without point cloud processing and move in a straight line?

lakshmip001, 2019-02-15:
> This does not use moveit. You can add a velocity cost or constraint to specific values for specific time steps. You can currently set both joint and cartesian velocities without time [...]

According to the example provided in #51, it converts the planning_response.trajectory of tesseract to trajectory_msgs using trajArrayToJointTrajectoryMsg() in traj_utilis.cpp assuming the last column obtained is time_from_start as shown below:

// Seperate out the time data in the last column from the joint position data
  auto pos_mat = traj_array.leftCols(traj_array.cols()-1);
auto time_mat = traj_array.rightCols(1);

But according to the coversions.cpp of trajopt it shows trajectory generated consists only joint_values of DOF of arm. So, time_from_start should be assigned by us I think. Kindly please correct me if I am wrong.

lakshmip001, 2019-02-20:
> I don't think I fully follow your question. [Here](https://github.com/mpowelson/industrial_training/blob/d2772749626da852cfb71eb31d1cb47ecc62ccb9/exercises/Optimization_Based_Planning/solution_ws/src/pick_and_place/src/test_bed_core_node.cpp#L212) is an example of using the TesseractPlanning Interface, but I don't think that will solve your problem.
> 
> It sounds like you're optimizing but not seeing anything in RVIZ? I've sometimes had to toggle the topic on and off or manually select the topic name. You might try that. As long as plotting_=true you should be seeing things with the examples. You might echo the topic and make sure something is showing up each time you press enter. If it does, just start clicking things in RVIZ until it shows up.

I would like to add linear motion to move from one point to the other keeping rotation fixed . In example above I see a funtion addlinearmotion(approach_pose, end_pose, steps) as shown below but I see change in angle for each step when remove changing angle it doesn't get acceleration and velocities .

void traj_prob::addLinearMotion(trajopt::ProblemConstructionInfo& pci,
                                Isometry3d start_pose,
                                Isometry3d end_pose,
                                int num_steps,
                                int first_time_step)
{
    Vector3d xyz_delta = (end_pose.translation() - start_pose.translation()) / (num_steps - 1);

    Quaterniond approach_rotation(start_pose.linear());
    Matrix3d rotation_diff = (start_pose.linear().inverse() * end_pose.linear());
    AngleAxisd aa_rotation_diff(rotation_diff);
    double angle_delta = aa_rotation_diff.angle() / (num_steps - 1);
    Vector3d delta_axis = aa_rotation_diff.axis();

    for(int i=0; i<num_steps; i++)
    {
        std::shared_ptr<CartPoseTermInfo> pose_constraint = std::shared_ptr<CartPoseTermInfo>(new CartPoseTermInfo);
            pose_constraint->term_type = TT_CNT;
            pose_constraint->link = ee_link_;
            pose_constraint->timestep = i + first_time_step;
            pose_constraint->xyz = start_pose.translation() + xyz_delta * i;

            Quaterniond rotation_delta(cos(0.5 * angle_delta * i),
                                       delta_axis.x() * sin(0.5 * angle_delta * i),
                                       delta_axis.y() * sin(0.5 * angle_delta * i),
                                       delta_axis.z() * sin(0.5 * angle_delta * i));
            Quaterniond rotation = rotation_delta * approach_rotation;

            /* Fill Code:
                 . Set the pose rotation
                 . Set pos_coeffs to all 10s
                 . Set rot_coeffs to all 10s
                 . Define the pose name as pose_[timestep]
                 . pushback the constraint to cnt_infos
            */
            /* ========  ENTER CODE HERE ======== */
            pose_constraint->wxyz = Vector4d(rotation.w(), rotation.x(), rotation.y(), rotation.z());
            pose_constraint->pos_coeffs = Vector3d(10.0, 10.0, 10.0);
            pose_constraint->rot_coeffs = Vector3d(0.0, 0.0, 0.0);
            pose_constraint->name = "pose_" + std::to_string(i + first_time_step);
            pci.cnt_infos.push_back(pose_constraint);

Can you help me understand how I can use for moving from one point to another in a linear motion.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson