tesseract-robotics/trajopt #60: convex solver failed! set TRAJOPT_LOG_THRESH=DEBUG to see solver output. saving model to /tmp/fail.lp and IIS to /tmp/fail.ilp

lakshmip001, 2019-01-04:
Hi, I get this error when I give there cartesian poses in a straight line motion . Kindly please help me how to diagnose this problem. Is there someway to input cartesian waypoints for the optimization to be converged as I dont have any obstacles in my environment and trying to test through rviz and cartesian poses are with respect to world? Is there any specific way to give cartesian poses for the optimized trajectory to converge or just three poses in a straight line can result in convergence? As I face issue with conergence error , please help me how I can solve and ask the robot to move.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson

jianye0428, 2020-03-02:
hello,
did you figure this problem?
could you please tell  me, how can this problem be fixed ?  I also get the same problem.

thank you in advance
Jian