tesseract-robotics/trajopt #62: Linking motion planning with robot drivers

lakshmip001, 2019-01-07:
I am able to see motion on rviz alone. But when I try connecting with the robot driver and running I dont see robot moving in both rviz and on the robot.

mpowelson, 2019-01-07:
This will depend on the driver you are using. I'm not sure about gazebo, but in general you probably need to put the output trajectory into a ROS trajectory message and either publish that to some topic or call some action. [Here](https://github.com/mpowelson/industrial_training/blob/d2772749626da852cfb71eb31d1cb47ecc62ccb9/exercises/Optimization_Based_Planning/solution_ws/src/pick_and_place/src/test_bed_core_node.cpp#L321) is a mediocre at best example for the KUKA iiwa.

lakshmip001, 2019-01-07:
I will check out example too. I am working with universal robot . If you have any example linking ur_modern_driver, please let me know.

lakshmip001, 2019-01-10:
> This will depend on the driver you are using. I'm not sure about gazebo, but in general you probably need to put the output trajectory into a ROS trajectory message and either publish that to some [...]

I did similar to that one above example , I am to see the trajectory using /trajopt/display_tesseract_trajectory but when I do the execute block it doesnt execute in the rviz and on the robot, it just displays the trajectory.Kindly please help me how I can move the robot arm using tesseract_planning on universal robot.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson