tesseract-robotics/trajopt #65: Trajectories generated dosent have joint velocities and joint acceleration

lakshmip001, 2019-01-11:
I created a trajectory and published to the joint_trajectory_action but the joint velocities and constraints are not available in the trajectory so my robot is not moving this the trajectory message being published.
positions: [-1.5295043600590792, -0.4678241770981601, -2.798475681586013, 1.6710705972247073, -1.5971836662472205, -1.53520336321362]
        velocities: []
        accelerations: []
        effort: []
        time_from_start: 
          secs: 4
          nsecs: 750000000
Kindly please let me know how I can solve this problem(add joint velocities in the trajectory)

gavanderhoorn, 2019-01-11:
Related: https://github.com/ros-industrial/ur_modern_driver/issues/253.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson