tesseract-robotics/trajopt #66: Error: "Hit enter key to step optimization!"  when launching trajopt_examples

hmz-15, 2019-01-19:
I have just built the packages and dependencies, and tried the examples in package "trajopt_examples". However, whichever launch file I used, the same error came out:

[ERROR] [1547913499.813080944]: Hit enter key to step optimization!

It seems that I didn't even get into the main function. It seems to be a quite simple problem, but I really can't find a way to deal with it. I appreciate it a lot for your help.

Levi-Armstrong, 2019-01-20:
Set the launch parameter plotting equal to false.

Levi-Armstrong, 2019-01-20:
`<arg name="plotting" default="true"/>`

mpowelson, 2019-01-21:
Just to clarify, the previous "error" is not actually an error. When you select plotting = true the optimization steps through each iteration one at a time so that you can see it. It is an error message simply so that it will be red, and you will notice it.

The second error is discussed in #56 . That actually is an error but it occurs after the path planning has occurred. It shouldn't effect you using it. Hang in there with these examples. It's on our to do list to clean these up, and we have a full tutorial in the works to add to the ROS-I training material soon.

aykutonol, 2019-05-29:
I had the exact same issue and resolved it by (re)installing libglfw by `sudo apt install libglfw3-dev libglfw3`. I hope that helps!

mpowelson, 2019-09-20:
The training mentioned above was published on the industrial training page. Find it here
https://industrial-training-master.readthedocs.io/en/melodic/_source/demo3/index.html

I believe the library unloader issue has been resolved as well now. However if you are still having issues, feel free to reopen this

closed 2019-09-20 by mpowelson