tesseract-robotics/trajopt #67: Interest in moving the optimization interface to IFOPT, COIN-OR, CASADI, or something els?

Levi-Armstrong, 2019-01-21:
There has been some internal discussion about if TrajOpt should adopt an existing optimization interface instead of the current one. There are several available that would expose additional optimization libraries that are not currently available. I wanted to get the current users feedback related to transitioning to an available optimization interface library like IFOPT, COIN-OR, CASSADI, ect? @gavanderhoorn @arocchi

arocchi, 2019-01-21:
In my opinion it would be very convenient to use an interface to be able to quickly switch between solvers. Unfortunately I don't know any that interfaces to the solvers we would be interested in having (qpOASES, ipOPT, OSQP, snopt, gurobi). On the other side, my opinion is that we would gain a huge benefit from using a robust non-linear optimization solver to tackle the whole optimization problem (e.g. CASADI) instead of using a hand-written nonlinear solver which then uses a robust QP solver underneath. The optimization method as described in the paper is IMHO good for a publication, but not state of the art or robust enough for real-world applications.

Levi-Armstrong, 2019-01-21:
It looks like currently CASADI currently support IPOPT, qpOASES, SNOPT out of the list but is missing OSQP and GUROBI. Looking at the code we have two options for the missing ones. We could write an interface and try and push it back and if that fails we could keep the interface in this repository. Do you see any issues with this approach?

Also is there any other packages that would be a good replacement?

mpowelson, 2019-01-21:
Levi,
I looked at [RobOptim](http://roboptim.net/) again. They list

Supported
- CFSQP
- IPOPT
- CMinPack
- NAG

Experimental
 - Eigen (Levenberg-Marquardt)
 - CMA-ES
- LNopt
- PaGMO
- KNITRO

Obviously this is missing qpOASES, SNOPT, and Gurobi, but they also have some robotics utilities that might be useful like their capsule package that generates convex hulls.

arocchi, 2019-01-21:
RobOptim used to be a promising project but I have the feeling is not being developed anymore and is left half-unfinished.

Levi-Armstrong, 2020-12-12:
We decided to move forward with IFOPT

closed 2020-12-12 by Levi-Armstrong