tesseract-robotics/trajopt #69: No module named git

lakshmip001, 2019-01-25:
sphinx-build . output
Running Sphinx v1.3.6
making output directory...

Exception occurred:
  File "conf.py", line 20, in <module>
    import git
ImportError: No module named 'git'
The full traceback has been saved in /tmp/sphinx-err-jyvjro9c.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!

tsijs, 2019-01-25:
I had the same issue. Apparently, I missed some dependency for the documentation generation to run. 
What did the trick for me was:

pip install GitPython

And I also missed recommonmark, which it complained about a second later:
pip install recommonmark

After this, the documentation was generated without errors, though it was practically empty. Is there practically no documentation for the current version or am I doing something wrong?

Levi-Armstrong, 2019-01-26:
It is mostly empty, the examples are the best source of information at the moment.

lakshmip001, 2019-01-28:
I installed git using 
pip install GitPython
still there is an error
lakshmi@lakshmi-Nitro-AN515-51:~$ pip install GitPython
Collecting GitPython
  Downloading https://files.pythonhosted.org/packages/fe/e5/fafe827507644c32d6dc553a1c435cdf882e0c28918a5bab29f7fbebfb70/GitPython-2.1.11-py2.py3-none-any.whl (448kB)
    100% |████████████████████████████████| 450kB 1.1MB/s 
Collecting gitdb2>=2.0.0 (from GitPython)
  Downloading https://files.pythonhosted.org/packages/da/30/a407568aa8d8f25db817cf50121a958722f3fc5f87e3a6fba1f40c0633e3/gitdb2-2.0.5-py2.py3-none-any.whl (62kB)
    100% |████████████████████████████████| 71kB 3.9MB/s 
Collecting smmap2>=2.0.0 (from gitdb2>=2.0.0->GitPython)
  Downloading https://files.pythonhosted.org/packages/55/d2/866d45e3a121ee15a1dc013824d58072fd5c7799c9c34d01378eb262ca8f/smmap2-2.0.5-py2.py3-none-any.whl
Installing collected packages: smmap2, gitdb2, GitPython
Successfully installed GitPython-2.1.11 gitdb2-2.0.5 smmap2-2.0.5
You are using pip version 8.1.1, however version 19.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
lakshmi@lakshmi-Nitro-AN515-51:~$ cd catkin_ws/
lakshmi@lakshmi-Nitro-AN515-51:~/catkin_ws$ cd src
lakshmi@lakshmi-Nitro-AN515-51:~/catkin_ws/src$ cd trajopt_ros/
lakshmi@lakshmi-Nitro-AN515-51:~/catkin_ws/src/trajopt_ros$ ghostscript 
GPL Ghostscript 9.26 (2018-11-20)
Copyright (C) 2018 Artifex Software, Inc.  All rights reserved.
This software comes with NO WARRANTY: see the file PUBLIC for details.
GS>^C
lakshmi@lakshmi-Nitro-AN515-51:~/catkin_ws/src/trajopt_ros$ cd gh_pages/
lakshmi@lakshmi-Nitro-AN515-51:~/catkin_ws/src/trajopt_ros/gh_pages$ sphinx-build . output
Running Sphinx v1.3.6

Exception occurred:
  File "conf.py", line 20, in <module>
    import git
ImportError: No module named 'git'
The full traceback has been saved in /tmp/sphinx-err-ajkeote0.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!

lakshmip001, 2019-01-28:
I solved by installing using following command
sudo apt-get install python3-git

But I get following error
Extension error:
Could not import extension sphinx.ext.githubpages (exception: No module named 'sphinx.ext.githubpages')

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson