tesseract-robotics/trajopt #70: AlignedUnorderedMap not in namespace of tesseract

lakshmip001, 2019-01-28:
/home/lakshmi/catkin_ws/src/trajopt_ros/trajopt/include/trajopt/utils.hpp:116:14: error: ‘AlignedUnorderedMap’ in namespace ‘tesseract’ does not name a template type
   tesseract::AlignedUnorderedMap<std::string, Eigen::Vector2d> pair_lookup_table_;
              ^
/home/lakshmi/catkin_ws/src/trajopt_ros/trajopt/include/trajopt/utils.hpp: In member function ‘void trajopt::SafetyMarginData::SetPairSafetyMarginData(const string&, const string&, const double&, const double&)’:
/home/lakshmi/catkin_ws/src/trajopt_ros/trajopt/include/trajopt/utils.hpp:80:5: error: ‘pair_lookup_table_’ was not declared in this scope
     pair_lookup_table_[obj1 + obj2] = data;
     ^
/home/lakshmi/catkin_ws/src/trajopt_ros/trajopt/include/trajopt/utils.hpp: In member function ‘const Vector2d& trajopt::SafetyMarginData::getPairSafetyMarginData(const string&, const string&) const’:
/home/lakshmi/catkin_ws/src/trajopt_ros/trajopt/include/trajopt/utils.hpp:92:15: error: ‘pair_lookup_table_’ was not declared in this scope
     auto it = pair_lookup_table_.find(key);

gavanderhoorn, 2019-01-28:
@lakshmip001: might be good if you could provide some context here instead of just a copy-paste of an error message.

lakshmip001, 2019-01-28:
actually I did sphinx-build . output
after that I am just doing catkin build to build the packages and this error occurs with this trajopt_ros in utils.hpp

Levi-Armstrong, 2020-12-12:
Reopen if needed.

closed 2020-12-12 by Levi-Armstrong