tesseract-robotics/trajopt #76: MoveIt packages

tsijs, 2019-02-14:
First of all, thank you all for making this package open-source. It seems to be quite useful for a project I am currently working on. 
I have a question however considering the moveit packages that are ignored, but nonetheless added to this repo. I didn't see it on the TODO list, but I couldn't help but wonder if the moveit package should work with the current implementation and if not, if it would be a lot of work (or even possible) to make it work.

tsijs, 2019-04-12:
I would really appreciate if someone could comment briefly on this. This hangs together now with the thread here: https://github.com/ros-planning/moveit/issues/1427

Levi-Armstrong, 2019-04-14:
Right now there are no plans to get the get the moveit plugin functional. I will be posting an action item list hopefully this week on one of the movie's open  issues outlining things that need to be addressed. Some pertain to moveit but there are other packages outside of moveit that also need to be modified, like urdf, geometric shapes, etc..

tsijs, 2019-04-22:
Thanks a lot for the clarification. This helps a lot. Since the moveit plugin seems to let to wait on itself for a few months, I am now trying to "hack" trajopt into a moveit plugin. 
I'd understand if you don't have time to support this, but if you could may be quickly give your opinion on the following, that would potentially save me from spending a lot of time on a dead end. 

tldr: can I use the collision checking plugin in the trajopt_moveit package instead of the collision manager classes to get the gradient and collision information?

What I am planning to do/ have done so far:
1. I used your bullet3 collision library plugin for moveit and changed the moveit API in my fork of moveit to incorporate the extra properties of the distance data for the continuous collision checking. (I tested the discrete checker which worked just out of the box, but didn't check the continuous one yet though)
2. I also replaced the collision checking managers content to call the collision plugin, which seems to pass back a similar struct with the same information the managers do. 
3. Copy the kdl kinematics classes and interface them with the problem definition made in 4. 
4. Put a problem definition in a moveit planning plugin.

I am now somewhere in between step 3 and 4. 4 is not that much work, but 3 gives me some headaches with dependencies to tesseract that I want to get rid of. Before spending too much time on this, I wanted to ask if this has any chance of succeeding ever or if there is a fundamental block that I don't see which will prevent this from ever working properly. 

Many thanks in advance!

Levi-Armstrong, 2020-12-12:
Reopen is needed.

closed 2020-12-12 by Levi-Armstrong