tesseract-robotics/trajopt #77: Coeffs

tsijs, 2019-02-14:
What do the pos_coeff & rot_coeff values represent in the problem definition files? 
I read through the code and it seems that they are penalties included in the cost function, but then why are they initialized to a value > 0 in the examples? It seems strange to penalize the position when it is different from the initialization. 
Or is this done for some kind of regularization?

mpowelson, 2019-02-14:
In general, the coeffs in the costs are just used to scale the costs relative to each other. If you have 2 joint velocity costs, one with coeffs of 10 and one with coeffs of 1, the first cost will be 10x as big as the second for the same velocity.

In the case of the cartesian we have weights on the rotation and the position. Granted, sending different values for each of the rotatation coeffs might give sort of unintuitive behavior. However, the example is showing that we can set the position coeff to 0 (essentially turning it off) while maintaining the orientation. Thus the "glass upright" example can move the middle points while keeping the orientation of the "glass" the same.

tsijs, 2019-02-14:
Thanks for the fast response! 

So if I understand it correctly:
In the json files, when the initial trajectory is defined, when putting a coeff > 0, will this incur cost when deviating from the initial position? Because in that case, the start and end-position of a to-be-planned trajectory should be defined in this config file and not in the node I think?
Though by playing with the example and changing the start/end position defined in the node main function, it seems to have an effect on the outcome of the optimization. 

Can you please clarify what the relationship between the initialized trajectory and the chosen start/end positions in the node main is?

mpowelson, 2019-02-14:
The initialized trajectory is just the initial conditions that are fed into the optimization. If you have start_fixed set, then the first point is set as fixed, but I don't think the examples have that. 

The cartesian costs are costs on the error from the target (which is usually different from the initial conditions). In many applications you would set a constraint on the initial conditions (b/c that's where the robot currently is) and a cost or constraint on the final position (b/c that's where you want to go) and then let the optimizer solve the middle ones. You might add costs to joint velocities and accelerations for all time steps b/c you want smooth trajectories. The same is true for collision costs

tsijs, 2019-02-15:
Ah, it becomes clear to me. I mistook the constraints for initialization. If I had only just read properly:p Thank you!

closed 2019-02-15 by tsijs