tesseract-robotics/trajopt #80: TrajArray.data in trajopt

lakshmip001, 2019-02-15:
I got the optimized trajectory for the robot arm of 6DOF with 6 values which I suppose the Joint angle values. So, the planning_response.trajectory doesnt contain time_from_start is it ? Or if I am wrong please correct me.

mpowelson, 2019-02-15:
If `pci.basic_info.use_time = false` then there is no time in the result. If it is true, then the last column will be 1/dt. However, the only costs that can use time (by setting TT_USE_TIME) at the moment is joint_velocity and a cost on total time.

mpowelson, 2019-02-15:
Time parameterization is still very much in development. I would recommend sticking with the optimizing the kinematics for now and then post process to get time.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson