tesseract-robotics/trajopt #82: Trajopt optimization doesn't move to correct end position

lakshmip001, 2019-02-21:
I give a start_pose and end_pose for optimizing the trajectory using trajopt. After I execute the joint trajectory, I see that it doesnt reach the end_pose with the following code

Vector3d xyz_delta = (end_pose.translation() - start_pose.translation()) / (num_steps - 1);

    Quaterniond approach_rotation(start_pose.linear());
    Matrix3d rotation_diff = (start_pose.linear().inverse() * end_pose.linear());
    AngleAxisd aa_rotation_diff(rotation_diff);
    double angle_delta = aa_rotation_diff.angle() / (num_steps - 1);
    Vector3d delta_axis = aa_rotation_diff.axis();

    for(int i=0; i<num_steps; i++)
    {
        std::shared_ptr<CartPoseTermInfo> pose_constraint = std::shared_ptr<CartPoseTermInfo>(new CartPoseTermInfo);
            pose_constraint->term_type = TT_CNT;
            pose_constraint->link = ee_link_;
            pose_constraint->timestep = i + first_time_step;
            pose_constraint->xyz = start_pose.translation() + xyz_delta * i;

            Quaterniond rotation_delta(cos(0.5 * angle_delta * i),
                                       delta_axis.x() * sin(0.5 * angle_delta * i),
                                       delta_axis.y() * sin(0.5 * angle_delta * i),
                                       delta_axis.z() * sin(0.5 * angle_delta * i));
            Quaterniond rotation = rotation_delta * approach_rotation;

            /* Fill Code:
                 . Set the pose rotation
                 . Set pos_coeffs to all 10s
                 . Set rot_coeffs to all 10s
                 . Define the pose name as pose_[timestep]
                 . pushback the constraint to cnt_infos
            */
            /* ========  ENTER CODE HERE ======== */
            pose_constraint->wxyz = Vector4d(rotation.w(), rotation.x(), rotation.y(), rotation.z());
            pose_constraint->pos_coeffs = Vector3d(10.0, 10.0, 10.0);
            pose_constraint->rot_coeffs = Vector3d(10.0, 10.0, 10.0);
            pose_constraint->name = "pose_" + std::to_string(i + first_time_step);
            pci.cnt_infos.push_back(pose_constraint);
    }
}
Kindly please let me know how to change the path to reach the end_pose.

mpowelson, 2019-02-21:
Where does it go instead of end_pose? This should work. It looks like it was taken from the example I sent you on another thread. It at least appears to work when we use it there. Are you saying our targets are in the wrong location or that the optimization is failing in some way, and you are getting incorrect results?

lakshmip001, 2019-02-22:
Actually optimization is working only the joints are getting interchanged before IPTP and after IPTP. I thought optimization was going wrong because of xyz_delta, rotation_delta,   pos_coeffs, rot_coeffs.

lakshmip001, 2019-02-22:
Actually I am able to solve my position problem but tool end-effector rotation I am unable to solve . I am giving two pose with same orientation when position my orientation same as the given and give to move my orientation changes from the starting point to the end point. I gave rot_coeffs as zero to prevent rotation still I have a problem.

mpowelson, 2019-02-22:
Setting the coffee to 0 effectively turns it off. Coeff is used to scale the cost. So you're scaling it to 0

lakshmip001, 2019-02-22:
I understood that setting rot_coeffs to some value doesn't constraint the motion and can easily rotate the end effector orelse if it is zero rotation is restricted . Kindly please correct if my understanding is wrong.

mpowelson, 2019-02-22:
This is backwards. Set coeffs to some value to add a cost to it. Set it to zero to let it rotate.

lakshmip001, 2019-02-22:
ok. How can I achieve straight line motion from one point to another point. How can I vary this pos_coeffs and rot_coeffs?

mpowelson, 2019-02-22:
Set them to some value. If all you have is Cartesian costs it doesn't really matter what it is. Try 1.

lakshmip001, 2019-02-22:
ok. I am using moveit time parameterization but unable to achieve trapezoidal cartesian velocity. How can achieve it and which are the costs I need to vary to achieve cartesian trapezoidal velocity?

gavanderhoorn, 2019-02-22:
@lakshmip001: observing this discussion I have a feeling that you're misinterpreting what `trajopt_ros`/`tesseract` do. `trajopt` is an optimisation library that happens to include robot motion planning domain concepts (ie: it "knows" what a joint pose is, what velocity limits are, etc).

Trapezoidal velocity profiles are not things that typically happen to be the outcomes of optimisation processes.

The fact that IPTP doesn't use/adhere to such profiles is known. I don't believe it claims anywhere to use them.

I'm not saying that what you want is wrong in any way, but the chances of an `n`D optimisation process outputting a trajectory with exactly a trapezoidal velocity profile are rather slim, due to the nature of how they work.

lakshmip001, 2019-02-22:
@gavanderhoorn optimization of trajectory with pose constraint costs and cartvelterminfo() cost affects the velocity trajectory profile .Correct me if I am wrong.
@Levi-Armstrong mentioned can try achieving trapezoidal velocity profile with three points giving a cartvelterminfo() cost for the end points as zero and some velocity for the middle point. But I am unable to give velocity cost for each point. 
Please correct me if I am wrong.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson