tesseract-robotics/trajopt #86: vel_lower_limit & vel_upper_limit for Universal Robot

lakshmip001, 2019-02-26:
// Taken from iiwa documentation (radians/s) and scaled by 0.8
      std::vector<double> vel_lower_lim{ 1.71*-0.8, 1.71*-0.8, 1.75*-0.8, 2.27*-0.8, 2.44*-0.8, 2.14*-0.8 };
      std::vector<double> vel_upper_lim{ 1.71*0.8, 1.71*0.8, 1.75*0.8, 2.27*0.8, 2.44*0.8, 2.14*0.8 };

are this values same for all robot or how does it affect the velocity of the universal robot?

mpowelson, 2019-02-26:
These are the velocity limits for the KUKA iiwa 7kg. They are dependent on the robot hardware. You would need to look up the values for a UR arm from the documentation. Or if you contact UR support, I'm sure they could give them to you.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson