tesseract-robotics/trajopt #87: Relating cartesian velocity obtained with the joint velocity and other parameters

lakshmip001, 2019-02-27:
I am trying to relate cartesian velocity with the joint velocities(jv) and CartVelTermInfo from the following code:
      /* ========  ENTER CODE HERE ======== */
      jv->targets = std::vector<double>(6, 0.0);
      jv->coeffs = std::vector<double>(6, 5.0);
    //  jv->lower_tols = vel_lower_lim;
    //  jv->upper_tols = vel_upper_lim;
      jv->term_type = TT_COST;
      jv->first_step = 0;
      jv->last_step = pci.basic_info.n_steps - 1;
      jv->name = "joint_velocity_cost";

      pci.cost_infos.push_back(jv);

      std::shared_ptr<CartVelTermInfo> cv(new CartVelTermInfo);
      cv->term_type = TT_COST;
      cv->first_step = 0;
      cv->last_step = pci.basic_info.n_steps-1;
      cv->max_displacement = 0.001;
      cv->link = "tool0";

      pci.cost_infos.push_back(cv);

mpowelson, 2019-02-27:
See #88 . I am unsure what you are asking and what is not working with the question in its current form.

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson