tesseract-robotics/trajopt #91: Trajopt has a position error of 1cm and 7cm along x and y axis pose

lakshmip001, 2019-03-01:
I am moving to a cartesian pose using trajopt which results in an error of 1cm and 7cm along pose_x and pose_y . Can I solve this by increasing the costs in pos_coeffs and rot_coeffs. What are the maximum and minimum values for pos_coeffs and rot_coeffs. And to move to the exact position , do I need to vary other variables.

mpowelson, 2019-03-04:
There are no max and min to the coeffs. They just scale the cost. I suppose at some point the cost will overflow if you make it too big, but I'd say as long as you're reasonable it should be fine. Unless you write some custom cost with extreme units, I'd say anything above 100 and I'd be suspicious something else is wrong. In my experience when the problem is set up well (no constraints infeasible) the solution has usually converged to something pretty intuitive regardless of the coeffs.

lakshmip001, 2019-03-04:
Actually solution gives an output that it is converged but when I see the position it reaches has a difference of 10cm along x-axis and 7cm y-axis. How can I make it to reach my end_pose with less error and if there would error in the position what can be the minimum error that I can minimise too. 

I feel the error is because trajopt tries to find the closestfeasiblesolution in a box and if it doesn't find trajopt increases the box size. If these is correct . There can be error existing in most of the end_poses which always doesn't reach its end point correctly. So there might be trajectory tracing to through the end_pose to move to the nextpoint .

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson