tesseract-robotics/trajopt #96: Should I define CartPoseTermInfo as constraint or costs

lakshmip001, 2019-03-06:
How does TT_CNT differ from TT_COST? Which is best to reach end_pose with no error? And when asked to move from start_pose to end_pose, it moves to a end_pose with little error and due to the movement back from end_pose to start_pose . At the starting robot moves in the opposite direction that it should move and then moves back to the start_pose with error. Can you please help me to understand why it moves in the opposite direction and later moves in the other direction?

gavanderhoorn, 2019-03-06:
@lakshmip001: have you had a chance to read the `trajopt` paper which explains the theoretical framework that is used? If not, please find it here: [Finding Locally Optimal, Collision-FreeTrajectories with Sequential Convex Optimization](http://joschu.net/docs/trajopt-paper.pdf).

That should get you familiar with the terminology, the mathematics and some of the trade-offs that go hand-in-hand with using a optimisation based approach to motion planning.

You might also want to take a look at the [trajopt documentation](http://rll.berkeley.edu/trajopt/doc/sphinx_build/html/) that explains how things are then used (note: this is Python, and the docs are for the (very) old version of `trajopt`, but they could still help in understanding the basic approach to problem setup and context configuration).

lakshmip001, 2019-03-06:
@gavanderhoorn thanks for the material . Actually I had this doubt because of the following issue

https://github.com/joschu/trajopt/issues/29

I tried using both cost and constraint . I gives an output that it is converged but there is error at end_pose or at the starting of the trajectory .

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson