tesseract-robotics/trajopt #119: raster_path_plan.launch example missing

blooop, 2019-07-31:
It looks like the ```trajopt_examples``` package has been moved to ```tesseract_ros``` but I can't find the raster example anywhere:
```roslaunch trajopt_examples raster_path_plan.launch```

The ```trajopt_examples``` package is still in ```trajopt_ros``` on the kinetic branch, but still doesn't have the raster example. 

Nothing comes up with a github search of:
```raster_path_plan in:path```

https://github.com/search?q=raster_path_plan&type=Code

Do you know where the example or something similar is?
Thanks

mpowelson, 2019-07-31:
That example was in a pull request that never got merged (#100). I have a work in progress PR [HERE](https://github.com/Levi-Armstrong/tesseract-1/pull/18) where I just copied it over, but it has not been fixed to work with the master branch.  However, taking the pick and place example, it should be pretty easy to deduce the necessary changes that need to be made.

blooop, 2019-07-31:
Thanks,  I just wasn't sure if there was a special approach used in that example.  I'll take a look and the PR and pick and place to reconstruct it.

closed 2019-07-31 by blooop