tesseract-robotics/trajopt #122: Depending on trajopt_ros from MoveIt

davetcoleman, 2019-08-07:
We've been making good progress on TrajOpt for MoveIt (thanks @ommmid!) and we are thinking about dependencies now. We're hoping we don't need to fork this repo, but rather share that planner code with Tesseract to prevent code duplication and share efforts in the future. What are your thoughts on this?

We need to depend on ``trajopt_sco`` and ``trajopt_utils`` from ``trajopt_ros``. However, ``trajopt_ros`` has another package, ``trajopt``, that is deeply dependent on ``tesseract``. As we do not want MoveIt to be dependent on Tesseract and only have ``trajopt_sco`` and ``trajopt_utils`` dependency, is it possible to move the ``trajopt`` package from this repo to the tesseract repo?  This way we can share changes in trajopt_ros with both MoveIt and Tesseract.

Thoughts?

Levi-Armstrong, 2019-08-07:
The trajopt package contains all of the cost and constraint functions along with the main problem description for defining the problem. The only things that is used from tesseract for trajopt is collision checking, forward kinematics and jacobian calculations. It may be worth exploring a generic interface or callbacks for these to break the dependency on tesseract. Also we will be rewriting trajopt in the very near future which MoveIt would benefit from as most of the work will be in the trajopt package.

ommmid, 2019-08-07:
That is actually what I am trying to do, trying to break the dependency on tesseract in trajopt package; adjusting kinematics_terms, problem_description and collision_terms so that they can read from MoveIt planning scene instead of tesseract env. I think I took care of JointPoseTemInfo constraint but CollisionTermInfo is more challenging, should we wait for the changes in trajopt as it is in the very near future? That would be great if new changes make the transition to MoveIt easier. Also, I will have much less time for trajopt at least for several weeks staring from next Friday.

mpowelson, 2019-08-07:
I think this goes back to something we have discussed internally several times internally. What exactly is Trajopt? Pragmatically, it is just a custom optimizing pathplanner at the moment. However, I think where we landed is that at least for us, the value added is in writing these costs and constraints for our types of problems and providing an interface to solvers that can solve them. With that in mind, the trajopt_sco package (basically the custom optimizer from the Schulman paper) is not really the core package. The trajopt package is the core package. That is also what is driving the rewrite we plan to do. We want to get rid of all of the custom optimization code and write the costs/constraints in a generic format (at the moment we are exploring IFOPT) that allows us to use off the shelf solvers written by people that do optimization for a living. Trajopt_sco would then be only one solver alongside things like IPOPT, SNOPT, Pagmo, etc.

As far as making a generic kinematics and collision interface, I would think that should not be that hard. That probably would be worth doing regardless.

davetcoleman, 2019-08-13:
> It may be worth exploring a generic interface or callbacks for these to break the dependency on tesseract

I think this is exactly what we would like to see for this repo, similar to how OMPL has no dependencies on MoveIt. 

Because @ommmid 's internship project is ending on Friday, unfortunately the current plan is to copy in the components of trajopt_ros into moveit. We would really appreciate help in removing these in the future so that moveit and tesseract both can share the future development effort of trajopt_ros repo.

mamoll, 2019-09-10:
@Levi-Armstrong @mpowelson has there been any progress on this from your side? Or are you waiting on PickNik to propose a solution?

mpowelson, 2019-09-19:
We have a couple of ideas we are working on. Hopefully we will flesh those out a bit more in the next week or so.