tesseract-robotics/trajopt #124: dependencies.rosinstall

WHGang, 2019-08-20:
After downloaded the package, i put it in my catkin workspace, there is a dependencies.rosinstall, How shold i use these file before building?
Could you list the commands that are needed to install trajopt_ros?

mpowelson, 2019-08-20:
That rosinstall file is mostly just there for continuous integration. You can read about how to run it [HERE](http://wiki.ros.org/wstool) if you want, but it is just going to pull Tesseract and its submodules. You might also want to run a rosdep on your workspace if you are still missing things. `rosdep install --from-paths src --ignore-src -r -y`

mpowelson, 2019-09-20:
I'm going to close this. If you are still needing more information or having the issue, feel free to reopen it.

closed 2019-09-20 by mpowelson