tesseract-robotics/trajopt #140: ContactTestType::ALL vs. ContactTestType::CLOSEST

schornakj, 2019-10-17:
Currently Trajopt's collision calculations are configured to return all objects in collision:

https://github.com/ros-industrial-consortium/trajopt_ros/blob/428405556047d1bff8c94888f1265aa6aca03925/trajopt/src/collision_terms.cpp#L323

This can cause optimization failures with numerous collision objects, such as a concave surface decomposed into many convex bodies, because the collision cost computed from these results isn't scaled by the number of objects in collision and is much higher than the cost from being near a smaller number of larger bodies.

A workaround is to change instances of `ContactTestType::ALL` to `ContactTestType::CLOSEST`, which just returns the single closest object in collision, so the collision cost is no longer directly dependent on the number of objects in collision.

Two thoughts:
- This would probably be a good thing to expose to the user.
- Would there be any disadvantage to switching the default from `ALL` to `CLOSEST`?

Levi-Armstrong, 2019-10-22:
Once we define the new interface this should be easy to accomplish.

Levi-Armstrong, 2020-01-02:
@schornakj  This was exposed in PR #152 .

closed 2020-01-02 by Levi-Armstrong