tesseract-robotics/trajopt #158: Issue building the examples

hzyjerry, 2020-01-22:
After building tesseract and trajopt_ros, I took trajopt examples from `kinetic-devel` [branch](https://github.com/ros-industrial-consortium/trajopt_ros/tree/kinetic-devel/trajopt_examples) and placed them in my catkin workspace.

However when doing `catkin build`, I ran into the following error
```
Errors     << trajopt_examples:cmake /home/jerrhe/catkin_ws/logs/trajopt_examples/build.cmake.000.log                                
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "trajopt" with any
  of the following names:

    trajoptConfig.cmake
    trajopt-config.cmake

  Add the installation prefix of "trajopt" to CMAKE_PREFIX_PATH or set
  "trajopt_DIR" to a directory containing one of the above files.  If
  "trajopt" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:6 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "trajopt" with any
  of the following names:

    trajoptConfig.cmake
    trajopt-config.cmake

  Add the installation prefix of "trajopt" to CMAKE_PREFIX_PATH or set
  "trajopt_DIR" to a directory containing one of the above files.  If
  "trajopt" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:6 (find_package)
```

Thank you for any insight!

Levi-Armstrong, 2020-01-22:
Is tesseract and trajopt in the same workspace?

hzyjerry, 2020-03-05:
@Levi-Armstrong thanks! Following up on this issue. Because the `trajopt_examples` module is from `kinetic-devel` branch which is a few hundred commits back, is there any recommended way to build it with the current master branch?

Thanks

Levi-Armstrong, 2020-03-05:
Most of these examples exist in the tesseract repository which can be used as a reference.

hzyjerry, 2020-03-06:
Perfect. I rebuilt the tesseract main repo and the examples in there work.

closed 2020-03-06 by hzyjerry