tesseract-robotics/trajopt #168: Return type of function doesn't match the variable type assigned while calling the function

skaruparthi, 2020-02-11:
In "trajopt_sco/src/gurobi_interface.cpp"
IntVec inds = vars2indsI(vars);

vars is of type sco::SizeTVec which is std::vector of type "long unsigned int"
IntVec is of type sco::IntVec which is std::vector of type "int"

This is causing the build to fail.

Levi-Armstrong, 2020-02-12:
Sorry, I do not have a license to Gurobi to confirm. If you created a PR to fix the issue we will get it merged into master.

skaruparthi, 2020-02-20:
I have made a couple of changes and got it working. How do I go about submitting a pull request? I tried the normal way but it just said permission denied. Sorry if it is a stupid question. Thanks in advance

Levi-Armstrong, 2020-02-21:
I would fork this repository and push your changes to your fork of this repository. Once that is done you then can create a PR from your fork to this repository.

Levi-Armstrong, 2020-05-27:
@skaruparthi Did #174 resolve your issue?

closed 2021-10-13 by Levi-Armstrong