tesseract-robotics/trajopt #183: Trajopt_ifopt TODO list

mpowelson, 2020-05-27:
- [ ] Collision - Add other collision evaluators
- [ ] Collision Callback - Collision is the only one that is missing
- [ ] Print debug info - trajopt_sco prints a bunch of debugging information that developers have become accustomed to. Some of this has been added to trajopt_sqp, but some like the cost associated with individual convex costs, will require some work to implement
- [ ] [Add Hessian to IFOPT](https://github.com/ethz-adrl/ifopt/issues/41)
- [ ] Look into adding caching to functions like Problem::EvaluateCostFunction

Levi-Armstrong, 2020-09-28:
@mpowelson Is this what is remaining for  completing trajopt_ifopt where it has the same functionality as trajopt?

mpowelson, 2020-09-28:
Big things I can think of are #192 and #193. Smaller issues, we are missing the dynamic cartesian constraint as well as joint acceleration. There are probably other things that will come up where you can semantically do the same thing but it looks a little different in the optimization (e.g. the old trajopt start_fixed flag), but I think the other collision evaluators are really the main thing that is missing before it will work for the majority of cases.

Of course time parameterization is also something that will need to be reworked.

colin-lewis-19, 2020-10-07:
I'm going to start chipping away at this, starting with #192

mpowelson, 2021-05-14:
Other possible performance improvements.

- [ ] Test out other linear solvers. OSQP supports [MKL Parsido].(https://osqp.org/docs/interfaces/linear_systems_solvers.html#linear-system-solvers-setting). "MKL Pardiso is an efficient multi-threaded linear system solver that works well for large scale problems part of the Intel Math Kernel Library"
- [ ] IFOPT problems don't change size, so we could use [OSQP code generation](https://osqp.org/docs/codegen/index.html) to generate a problem once and solve it repeatedly. I'm not sure if this is supposed to be faster or is just for deploying the QP solver in an external codebase

simonschmeisser, 2022-10-11:
Maybe I'm missing something obvious, but what's the difference between ifopt and sco? When would I use one or the other? Or is ifopt a successor to sco? If you provide me with some hints I can add a few lines to the readme

mpowelson, 2022-10-11:
Yes, trajopt_ifopt + trajopt_optimizers was intended as a successor to trajopt + trajopt_sco. It was intended to make the creation of the problem and the solver used more separate so we could more easily add other optimizers. Currently trajopt_ifopt is not recommended. We've used it for some examples, but there are still a few things to be worked out. However, there are some applications like [this online planning example](https://rosindustrial.org/news/2020/8/4/real-time-trajectory-planning-for-industrial-robots) which aren't really possible with the way the old trajopt was structured.

closed 2026-01-19 by Levi-Armstrong