tesseract-robotics/trajopt #187: Update CI

mpowelson, 2020-06-09:
For consistency we probably want to update to github actions. Additionally, this uses the legacy branch of industrial_ci. We should update it to use master. Noetic and foxy builds would be nice as well.

- [ ] Update to github actions
- [ ] Update to industrial CI master branch
- [ ] Add ROS 1 Noetic build
- [ ] Add ROS2 Foxy build

Minipada, 2020-07-07:
Hi from World ROS-I day

gavanderhoorn, 2020-07-07:
@Minipada: please note the comment by @mpowelson about using `industrial_ci` for the Github actions.

Minipada, 2020-07-07:
Thanks for the confirmation, I was taking a look at it

Minipada, 2020-07-07:
Hi, I first tested making it work with Melodic
short questions:
1. It seems there are circular dependencies. trajopt needs tesseract which needs in return trajopt. I don't see it as an issue but the CI will build it twice, once in the upstream source and once in the workspace. In the ci file, it will translate to having "github:ros-industrial-consortium/trajopt_ros#master" in the `UPSTREAM_WORKSPACE`. If this is fine to you, I'll keep it this way
2. Could you give me a link to the `lharmstrong/tesseract` dockerfiles?

mpowelson, 2020-07-07:
1) On Tesseract we solve this issue by building both Tesseract and Trajopt in the target workspace. The other packages in the upstream (descartes_light, opw_kinematics, and ifopt) do not need Tesseract, so it shouldn't have to be included twice.

2) I think you should be able to use the dockers being build by ros-industrial/docker now. @Levi-Armstrong 
 merged the Tesseract dockerfiles a few days ago. 

https://hub.docker.com/r/rosindustrial/tesseract/dockerfile

Levi-Armstrong, 2021-04-20:
This has been completed.

closed 2021-04-20 by Levi-Armstrong