tesseract-robotics/trajopt #192: Add collision callback to trajopt_sqp

mpowelson, 2020-07-06:
Address this.
https://github.com/ros-industrial-consortium/trajopt_ros/blob/9360773d3dd672e5d698ba9535d60fbd338348e1/trajopt_optimizers/trajopt_sqp/src/callbacks/collision_plotter.cpp#L39

mpowelson, 2020-07-06:
This should be relatively straightforward. See the plot functions in the old trajopt implementation. You can also look at how the other callbacks work in relation to their old trajopt implementation

https://github.com/ros-industrial-consortium/trajopt_ros/blob/9360773d3dd672e5d698ba9535d60fbd338348e1/trajopt/src/collision_terms.cpp#L985

Srujan-D, 2020-07-07:
Hey! I'd like to work with this issue!

cardboardcode, 2020-07-07:
Hi @gavanderhoorn, may I get your help in assigning @Srujan-D to this issue?

I am unable to do so due to lack of ownership for this particular repository.

gavanderhoorn, 2020-07-07:
I've created the team and added @Srujan-D. He/she will have to accept the invitation though before the issue can be assigned.

Srujan-D, 2020-07-07:
Yep, received and accepted the invite. Thanks for your help @cardboardcode and @gavanderhoorn !

mpowelson, 2020-07-07:
@Srujan-D If you need any help let us know. Thanks for working on this!

gavanderhoorn, 2020-07-07:
@mpowelson: he was in a time-zone which made it difficult to wait on you guys.

@Srujan-D then asked to try and work on something else.

mpowelson, 2020-07-07:
Ah. That's fine then.