tesseract-robotics/trajopt #203: trajoptConfig.cmake compiling issue

jingweirobot, 2020-09-23:
-- Could NOT find trajopt (missing: trajopt_DIR)
-- Could not find the required component 'trajopt'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "trajopt" with any
  of the following names:

    trajoptConfig.cmake
    trajopt-config.cmake

  Add the installation prefix of "trajopt" to CMAKE_PREFIX_PATH or set
  "trajopt_DIR" to a directory containing one of the above files.  If
  "trajopt" provides a separate development package or SDK, be sure it has
  been installed.

when I ran industrial-ros demo in the section 4. it shows such error. Could you help me fix this thing? Thanks in advance

Levi-Armstrong, 2020-09-23:
Could you provide a link to the demo in question?

Levi-Armstrong, 2021-04-20:
Please reopen if still an issue.

closed 2021-04-20 by Levi-Armstrong