tesseract-robotics/trajopt #211: trajopt_ros does not build with the latest version of tesseract_ext

haudren, 2020-10-19:
Hello,

I just noticed that ros-industrial-consortium/tesseract_ext#27 broke the build of `trajopt_tools`.  Indeed, it disables the build of bullet extras, but the `ConvexDecomposition` library is a bullet extra, and is required here:

https://github.com/ros-industrial-consortium/trajopt_ros/blob/c946474ebe86d9b889c0fe2a4add7b8bbfd385aa/trajopt_tools/CMakeLists.txt#L11

Steps to reproduce:
- Download latest `tesseract_ext` ( 1ad88ca ) and `trajopt_ros` ( c946474 )
- Build using `catkin build` (Ubuntu 18.04, ROS Melodic)
- Build errors out with:
```
Errors     << trajopt_tools:cmake /tmp/test_ws/logs/trajopt_tools/build.cmake.000.log
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CONVEX_DECOMPOSITION_LIBRARY
    linked by target "convex_decomposition_hacd_node" in directory /tmp/test_ws/src/trajopt_ros/trajopt_tools
HACD_LIBRARY
    linked by target "convex_decomposition_hacd_node" in directory /tmp/test_ws/src/trajopt_ros/trajopt_tools
```

Reverting to the previous commit on `tesseract_ext` fixes the issue. Unfortunately, I couldn't easily find a subset of flags that would allow to build Bullet without the examples but with the extras, as it seems like part of the extras depend on "-lBulletExampleBrowserLib".

haudren, 2020-10-19:
Side note, the other workaround would be to remove the `hacd` node from `trajopt_tools` and rely on `vhacd` only.

schornakj, 2020-10-19:
I've encountered this issue as well.

Levi-Armstrong, 2020-10-19:
I am good with enable external back in tesseract_ext it was an oversight on my part.

Levi-Armstrong, 2020-10-19:
Opened a [PR](https://github.com/ros-industrial-consortium/tesseract_ext/pull/28/files) to resolve the issue. When CI passes I will merge.

Levi-Armstrong, 2020-10-20:
The PR has been merged.

closed 2020-10-20 by Levi-Armstrong

hehonglu123, 2021-01-11:
I had the problem with latest tesseract_ext now, fixed by reverting to `7a8e5e52ef298fb39ba9090a26c6209421100332`.