tesseract-robotics/trajopt #221: BPMPD Thread Safety

marip8, 2020-12-09:
I tried solving an optimization using the `BPMPD` solver in a multi-threaded context and encountered a segfault deep inside `BPMPD`. Switching the solver to both `OSQP` and `qpOASES` solves the issue, which seems to indicate that some aspect of `trajopt` with the `BPMPD` solver is not thread safe. I'm not sure if this comes from the solver library itself or the `trajopt_sco` wrapper.

I encounter this issue on at [this version](b84e4ae30fa658d2911f3e375bb6009dd8ba7084) of the repository.

marip8, 2020-12-09:
If I had to hazard a guess, this issue may be caused by the couple of static variables in the `BPMPD` interface library that get used by multiple functions

https://github.com/ros-industrial-consortium/trajopt_ros/blob/b84e4ae30fa658d2911f3e375bb6009dd8ba7084/trajopt_sco/src/bpmpd_interface.cpp#L205-L207

@Levi-Armstrong @mpowelson is there a reason that these variables are static and not a member of the class?

Levi-Armstrong, 2020-12-09:
Nope, This was just how it was originally implemented. Change if needed.