tesseract-robotics/trajopt #237: Conflicts between trajopt and trajopt_ifopt in Collion terms

mpowelson, 2021-01-20:
Bad things happen when something is linked against both trajopt and trajopt_ifopt. The collision cost in trajopt_ifopt somehow ends up calling the CollisionEvaluator in trajopt. We probably ought to change the namespace of one of them. Since the original trajopt version is not used externally, I'd prefer we change that one.

Levi-Armstrong, 2021-01-20:
What would want to rename it? Though it may look odd that the trajopt_ifopt has the namespace trajopt and the trajopt package has a different namespace.

mpowelson, 2021-01-20:
I guess the most consistent thing to do would be put all of trajopt ifopt in the trajopt_ifopt namespace. Renaming the original collision evaluator to Trajopt collision evaluator might be enough though.

mpowelson, 2021-01-22:
Discussing with @Levi-Armstrong the consensus was we should just switch trajopt_ifopt to use the trajopt_ifopt namespace.