tesseract-robotics/trajopt #254: Error building the repo

mohamedsayed18, 2021-04-09:
I'm trying to build the repo, I did the following steps:
1. `mkdir -p traj_opt_space/src`
2. I cloned the repo, and other repos need like the ros_industrial boiler plate.
3. In the root of the work space, I write `catkin build`
Output:
```
Errors     << trajopt_ifopt:make /media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/logs/trajopt_ifopt/build.make.000.log              
/media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/src/trajopt_ros/trajopt_ifopt/src/squared_cost.cpp: In member function ‘virtual void trajopt::SquaredCost::FillJacobianBlock(std::__cxx11::string, ifopt::Component::Jacobian&) const’:
/media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/src/trajopt_ros/trajopt_ifopt/src/squared_cost.cpp:76:56: error: ‘virtual void ifopt::ConstraintSet::FillJacobianBlock(std::__cxx11::string, ifopt::Component::Jacobian&) const’ is private within this context
   constraint_->FillJacobianBlock(var_set, cnt_jac_block);
                                                        ^
In file included from /opt/ros/noetic/include/ifopt/cost_term.h:33,
                 from /media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/src/trajopt_ros/trajopt_ifopt/include/trajopt_ifopt/costs/squared_cost.h:31,
                 from /media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/src/trajopt_ros/trajopt_ifopt/src/squared_cost.cpp:26:
/opt/ros/noetic/include/ifopt/constraint_set.h:109:16: note: declared private here
   virtual void FillJacobianBlock(std::string var_set, Jacobian& jac_block) const = 0;
                ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/trajopt_ifopt.dir/build.make:141: CMakeFiles/trajopt_ifopt.dir/src/squared_cost.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/trajopt_ifopt.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/traj_opt_space/build/trajopt_ifopt; catkin build --get-env trajopt_ifopt | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
.............................................................................................................................................
Failed     << trajopt_ifopt:make                              [ Exited with code 2 ]                                                         
Failed    <<< trajopt_ifopt                                   [ 3 minutes and 57.9 seconds ]                                                 
Abandoned <<< trajopt_sqp                                     [ Unrelated job failed ]                                                       
[build] Summary: 19 of 21 packages succeeded.                                                                                                
[build]   Ignored:   None.                                                                                                                   
[build]   Warnings:  6 packages succeeded with warnings.                                                                                     
[build]   Abandoned: 1 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 15 minutes and 3.0 seconds total.                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

Levi-Armstrong, 2021-04-09:
A while back we had a PR against IFOPT exposing this function so you may have an older version. How did you install ifopt? Also what version of Ubuntu are you using?

mohamedsayed18, 2021-04-09:
> How did you install ifopt? 

`sudo apt install ros-noetic-ifopt`
>Also what version of Ubuntu are you using?

`Ubuntu 20.04.2 LTS`

Levi-Armstrong, 2021-04-09:
So I just checked and the [latest state](https://github.com/ethz-adrl/ifopt/issues/69) of IFOPT has not been release so you will need to clone it into your workspace.

mohamedsayed18, 2021-04-10:
Finally, I was able to build the repo. As you suggested in the beginning it was a problem with Eigen library
```
export EIGEN3_INCLUDE_DIR=/media/mohamed/C03CCDB43CCDA62E/work/Code/eigen-3.3.7/
export EIGEN_INCLUDE_DIR=/media/mohamed/C03CCDB43CCDA62E/work/Code/eigen-3.3.7/
```
I set these environment variables in my bashrc and it worked
Thank you very much for your continuous responses

closed 2021-04-10 by mohamedsayed18