tesseract-robotics/trajopt #295: Build Failure with GUROBI support

cpetersmeier, 2022-08-03:
Hello everyone,

I am getting a build failure when building `trajopt_sco` with gurobi support:

```
/workspace/src/trajopt/trajopt_sco/src/gurobi_interface.cpp: In function ‘void sco::resetIndices(sco::VarVector&)’:
/workspace/src/trajopt/trajopt_sco/src/gurobi_interface.cpp:160:20: error: no match for ‘operator[]’ (operand types are ‘sco::VarRep::Ptr’ {aka ‘std::shared_ptr<sco::VarRep>’} and ‘size_t’ {aka ‘long unsigned int’})
  160 |     vars[i].var_rep[i].index = i;
      |                    ^
/workspace/src/trajopt/trajopt_sco/src/gurobi_interface.cpp: In function ‘void sco::resetIndices(sco::CntVector&)’:
/workspace/src/trajopt/trajopt_sco/src/gurobi_interface.cpp:165:20: error: no match for ‘operator[]’ (operand types are ‘sco::CntRep::Ptr’ {aka ‘std::shared_ptr<sco::CntRep>’} and ‘size_t’ {aka ‘long unsigned int’})
  165 |     cnts[i].cnt_rep[i].index = i;
      |                    ^
make[2]: *** [CMakeFiles/trajopt_sco.dir/build.make:180: CMakeFiles/trajopt_sco.dir/src/gurobi_interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/trajopt_sco.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

I included `gurobi95` in this line:
https://github.com/tesseract-robotics/trajopt/blob/b53d75247a280092b3207bbd2e3ad21c77baa9ae/trajopt_sco/cmake/FindGUROBI.cmake#L21

Levi-Armstrong, 2022-08-09:
Sorry, I currently do not have a license to help with the build issues. If you work out the build issue create PR and I will merge it in.

Levi-Armstrong, 2022-08-09:
Looking at the other solvers, I believe these should be updated:
```c++
vars[i].var_rep->index = i;
cnts[i].cnt_rep->index = i;
```

cpetersmeier, 2022-08-10:
This looks promising! I was able to compile and run the application with the proposed changes and created a pull request here https://github.com/tesseract-robotics/trajopt/pull/296. Thanks so much!

closed 2022-09-16 by Levi-Armstrong