tesseract-robotics/trajopt #300: ROS release?

simonschmeisser, 2022-10-11:
I'm currently spending some time on integrating trajopt as a MoveIt planner and was wondering if there are plans for a release to the ros noetic build farm? I have seen tesseract has been released already.

simonschmeisser, 2022-10-11:
https://github.com/ros-planning/moveit/pull/3235 is my WIP branch for adding collision terms

simonschmeisser, 2022-10-12:
Yes that's another attractive approach, I'll try to implement that as well to see if I can reproduce your collision checking performance plot (or do you have the source somewhere?)

(tesseract_planning has not been released either, right?)

Levi-Armstrong, 2022-10-13:
It is not, mostly because some of the dependencies are not release like [osqp](http://repositories.ros.org/status_page/ros_noetic_default.html?q=osqp).  I reached out to @esteve to see about being added as maintainer to fix the issue. I think after this is release it should be possilble to release tesseract_planning and tesseract_ros.

simonschmeisser, 2023-01-22:
Hi Levi,
if I understood things correctly osqp has been released in the mean time?

Levi-Armstrong, 2023-01-22:
I believe so.