tesseract-robotics/trajopt #305: How to visualize a trajectory in rViz?

allicen, 2023-02-25:
When I run the example **roslaunch tesseract_ros_examples glass_upright_example.launch**, I only see a point cloud. And I can't start executing the TrajOpt trajectory in rViz:
[image]
Any ideas how to do this? Thank you

Levi-Armstrong, 2023-02-25:
What version are you using? In the tesseract panel it should have a trajectory tab which allows you to play, pause and select states.

Levi-Armstrong, 2023-02-25:
https://github.com/tesseract-robotics/tesseract_qt/blob/main/docs/images/tesseract_qt_joint_trajectory_widget.png

allicen, 2023-02-25:
Thanks! I use Ubuntu 20.04 + Ros Noetic + Docker.
This button is not visible in the default theme rViz:
[image]

I can't choose here. Do I need additional settings?
[image]

Levi-Armstrong, 2023-02-26:
Hit the refresh button which is the last toolbar button. It grabs a snap shot of the kinematics objects since in Tesseract these can be changed.

allicen, 2023-02-26:
Thanks, it works!

closed 2023-02-26 by allicen