tesseract-robotics/trajopt #312: Avoid recreating the problem with in the trust region loop

Levi-Armstrong, 2023-03-14:
Most if not all support just updating lower and upper bounds and resolving without recreating the problem. Though currently the problem is recreated and should be updated so this does not occur.

rjoomen, 2025-08-28:
Solved for trajopt_sco with osqp here: https://github.com/tesseract-robotics/trajopt/pull/446

rjoomen, 2025-08-28:
Suppose so, but that uses osqp-eigen, so it'll work a bit differently. But I'm working on upgrading osqp and osqp-eigen, so let's wait till that's finished.

Levi-Armstrong, 2025-09-03:
> And another thing is that the workspace updating doesn't work all that well for me. I'm seeing a lot of solver failures. With the bounds fix [#472](https://github.com/tesseract-robotics/trajopt/pull/472), at least the warm starting seems to work better, so that's good. I'm curious to see how the updating will work with osqp 1.0.0.

Should this be disabled for now?

Levi-Armstrong, 2025-09-06:
>Or could it be that Trajopt swaps a constraint with another and positions it on the same position in the matrix?

This is most likely happening with the collision constraint, but most of the other constraints should be deterministic.