tesseract-robotics/trajopt #325: Naming of CollisionCoeffData::setPairCollisionMarginData()

rjoomen, 2023-06-02:
I seems to me that [CollisionCoeffData::setPairCollisionMarginData()](https://github.com/tesseract-robotics/trajopt/blob/8066b980f35f245a68961b290d44cf1a46770ed2/trajopt_ifopt/src/constraints/collision/collision_types.cpp#L35) should be called CollisionCoeffData::setPairCollisionCoeff(), as it sets the coefficient instead of the margin (the getter is correctly called getPairCollisionCoeff()).

Should I submit a PR?

Levi-Armstrong, 2023-06-02:
You are correct. Sure please do.

rjoomen, 2023-06-05:
Same file: is [this line](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_ifopt/src/constraints/collision/collision_types.cpp#L9) still correct?

Levi-Armstrong, 2023-06-05:
I would say that line can be removed

rjoomen, 2023-06-06:
I removed it at a few more places, please review.

PR #326 (2023-06-06): Rename setPairCollisionMarginData to setPairCollisionCoeff
Fixes #325
closed 2023-06-06 by Levi-Armstrong