tesseract-robotics/trajopt #340: trajopt_ifopt does succeed if max_merit_coeff_increases is exceeded

rjoomen, 2023-07-28:
If during the penalty iteration loop the status_ is set to NLP_CONVERGED, but verifySQPSolverConvergence() still detects constraint violations, the loop continues, until max_merit_coeff_increases is reached. But because verifySQPSolverConvergence() does not set the status back to RUNNING, it can happen that solve() returns with status_ NLP_CONVERGED although constraints are still violated.

I suggest to either add status_ = SQPStatus::RUNNING; [here](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_optimizers/trajopt_sqp/src/trust_region_sqp_solver.cpp#L134), or add a new SQPStatus::PENALTY_ITERATION_LIMIT with a matching check.

Levi-Armstrong, 2023-07-30:
Great find. I went through the legacy trajopt and found a few other behaviors which are different. I will have a PR soon.

Levi-Armstrong, 2023-07-30:
This [PR](https://github.com/tesseract-robotics/trajopt/pull/341) should address this issue.

rjoomen, 2023-08-01:
Solved by #341, thanks!

closed 2023-08-01 by rjoomen