tesseract-robotics/trajopt #361: Discrete collision check setting

rjoomen, 2023-11-20:
Shouldn't `is_continuous` be false [here](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_ifopt/src/constraints/collision/continuous_collision_evaluators.cpp#L351), as it is the `LVSDiscreteCollisionEvaluator`?

rjoomen, 2023-11-20:
Otherwise `LVSContinuousCollisionEvaluator::CalcCollisionData()` and `LVSDiscreteCollisionEvaluator::CalcCollisionData()` are _exactly_ the same.

Levi-Armstrong, 2023-11-20:
It needs to be set to true, it is still considered continues from a trajopt perspective even though it is using a discrete contact checker because it is integrating collision data along the path between two state.

Levi-Armstrong, 2026-01-20:
I am onboard with figuring out a way to consolidate this.