tesseract-robotics/trajopt #370: TrajOptIfopt errors in getMaxErrorWithBufferT*

rjoomen, 2023-11-29:
The [TrajOptIfopt version of car_seat_example](https://github.com/tesseract-robotics/tesseract_planning/pull/389) crashes because ["Max error with buffer at **T0** does not exist."](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_common/src/collision_types.cpp#L237). This is using the LVS_DISCRETE collision evaluator. In testing with my own code I also have crashes, because ["Max error with buffer at **T1** does not exist."](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_common/src/collision_types.cpp#L257) when using evaluator CONTINUOUS, but not when using the other collision evaluator types.

I have been searching for hours, but can't find the reason. @Levi-Armstrong would you be able to take a look, because I've no idea how to solve this.

Levi-Armstrong, 2023-11-29:
Yea, I will take a look. What branch should I check out.

rjoomen, 2023-11-29:
[This one](https://github.com/rjoomen/tesseract_planning/tree/ifopt_examples), and then build [car_seat_example_node ](https://github.com/rjoomen/tesseract_planning/blob/ifopt_examples/tesseract_examples/src/car_seat_example_node.cpp#L45)with parameter ifopt = true and run it. It should quit with a non-descriptive error.

Levi-Armstrong, 2023-12-04:
Fixed by PR #372

closed 2023-12-04 by Levi-Armstrong

rjoomen, 2023-12-05:
Great, this has solved the issue. Thanks.