tesseract-robotics/trajopt #373: OSQP infinity

rjoomen, 2023-12-04:
It seems to be the OSQP Eigen interface (in SQP) should use [INFINITY instead of OSQP_INFTY](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_optimizers/trajopt_sqp/src/osqp_eigen_solver.cpp#L278C87-L278C87), as [SCO also uses that](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_sco/src/osqp_interface.cpp#L18C22-L18C22).

Otherwise the initial bounds do not match between Trajopt and TrajoptIfopt.

Levi-Armstrong, 2023-12-04:
I think this slight discrepancy should be fine. In my testing I did not see any difference in the output table data. Are you seeing anything significant?

rjoomen, 2023-12-04:
No, it does not seem to make a difference, but it shows in OSQP_COMPARE_DEBUG_MODE. I'll PR it sometime, if that's OK for you.

closed 2023-12-05 by rjoomen

Levi-Armstrong, 2023-12-05:
I have reviewed several issue posted on OSQP and I think it would be best to use OSQP_INFTY instead because it looks like can lead to failures like iteration limits, etc. See the discussion [here](https://github.com/osqp/OSQP.jl/pull/13). I would be in favor in updating the legacy version of using OSQP_INFTY if you want to make that change.

rjoomen, 2023-12-05:
But if we really always want to use OSQP_INFTY when OSQP is the solver we also need to deal with lines [like this](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_optimizers/trajopt_sqp/src/trajopt_qp_problem.cpp#L199) and [probably this](https://github.com/tesseract-robotics/trajopt/blob/89661ad2b2d02d7c1cdf3e1a765accd56f9d724f/trajopt_sco/src/modeling.cpp#L21). Although I did some quick testing and changing this did not change my results.

rjoomen, 2024-01-08:
I fixed it in `osqp_interface.cpp`.

Doing the same in `trajopt_qp_problem.cpp` and `modelling.cpp` is not possible though, as the type of solver is not known at these points, so other infinity limits might apply, depending on the solver. As the current infinity does not seem to create problems, I'll leave it at this.