tesseract-robotics/trajopt #384: CMake issue with `jsoncpp_lib` ALIAS target

marip8, 2024-01-30:
The following line fails for me with CMake 3.16.3, resulting in the error: `add_library cannot create ALIAS target "jsoncpp_lib" because target "JsonCpp::JsonCpp" is imported but not globally visible.`

https://github.com/tesseract-robotics/trajopt/blob/ac685dc2fc8d7050d54a7b6fce40422374f81ff5/trajopt_sco/cmake/trajopt_sco-config.cmake.in#L15

This error seems to go away with CMake >= 3.25.0 (possibly with an earlier version). This also corresponds to [this reported issue](https://discourse.cmake.org/t/alias-and-not-globally-visible-targets/6305).

I think the solution should be to update the minimum CMake version of all packages that perform the operation in the code snippet above (`trajopt_sco`, `trajopt`) to CMake >= 3.25.0

rjoomen, 2024-01-30:
See also #298

Levi-Armstrong, 2024-02-07:
What impact does this have on those running 20.04?

marip8, 2024-02-08:
I encountered this issue on 20.04. Presumably anyone running CMake <= 3.25.0 on any supported distro will run into this