tesseract-robotics/trajopt #402: Default planning time

allicen, 2024-05-01:
How can I change the planning time? Set more than 5 seconds

Thank you!
```
==================
Optimization results:
status: OPT_PENALTY_ITERATION_LIMIT
cost values: (2.90515e-26, 4.2114e-23, 0, 0, 0, 0.00376514, 0.250181, 2.09396, 1.40071, 0.78795, 0, 0, 0.642342)
constraint violations: (2.90515e-26, 4.2114e-23, 0, 0, 0, 0.00376514, 0.250181, 2.09396, 1.40071, 0.78795, 0, 0)
n func evals: 20
n qp solves: 21
==================
Info:    TrajOptMotionPlannerTask motion planning failed (Failed to find valid solution: OPT_PENALTY_ITERATION_LIMIT) for process input: Tesseract Composite Instruction
Debug:   Error
Info:    Planning took 5.572277 seconds.
```

marip8, 2024-05-01:
I don't think `trajopt` is timing out here. The specific error that you are seeing here is a "penalty iteration limit", which means that:
- the constraints of your optimization problem could not be satisfied
- `trajopt` increased the weight of their error (i.e., penalty) in the optimization by an order of magnitude
- the optimization still did not converge, so `trajopt` repeated this process the maximum number of times before concluding that the optimzation could not be solved

This whole process coincidentally took about 5 seconds,  but I don't think it technically timed out. You can change the penalty iteration limit in one of the `trajopt` profiles, but I doubt increasing it will solve your problem

Lots of things can cause constraints to not be satisfied, resulting in this error. If you could post the contents of the last iteration of the `trajopt` log with the list of costs/constraints and their values, we might be able to help you resolve this issue

allicen, 2024-05-05:
Thanks for the answer! This allowed me to better understand the situation.

I am attaching the logs.
[trajopt_constraints.log](https://github.com/tesseract-robotics/trajopt/files/15214354/trajopt_constraints.log)
[trajopt_costs.log](https://github.com/tesseract-robotics/trajopt/files/15214355/trajopt_costs.log)
[trajopt_vars.log](https://github.com/tesseract-robotics/trajopt/files/15214357/trajopt_vars.log)
[trajopt_solver.log](https://github.com/tesseract-robotics/trajopt/files/15214358/trajopt_solver.log)

Logs from the terminal: 

`Info:    UR5 trajopt plan
Debug:   generating program
Program: Composite Instruction, Description: Tesseract Composite Instruction
Program: {
Program:   Move Instruction, Move Type: 1, State WP: Pos=     1.52647    -0.553581      1.68679     -2.68775     -1.46159 -0.000419442
, Description: Start Instruction
Program:   Move Instruction, Move Type: 1, State WP: Pos=     1.54235     -1.20335     0.634284     -1.13847      -1.5738 -0.000371281
, Description: freespace_finish_plan
Program: }
[ INFO] [1714939743.165975455]: Planner task name - TrajOptPipeline
Debug:   Profile 'FREESPACE' was not found in namespace 'MinLengthTask' for type 'N18tesseract_planning16MinLengthProfileE'. Using default if available. Available profiles:
Debug:   Environment, getJointGroup(manipulator) cache miss!
Debug:   Environment, getGroupJointNames(manipulator) cache miss!
Debug:   distances and parents:
Debug:   distance(ur5_wrist_2_link) = 1.016989, parent(ur5_wrist_2_link) = ur5_wrist_1_link
Debug:   distance(gripper_finger2_knuckle_link) = 1.177339, parent(gripper_finger2_knuckle_link) = gripper_base_link
Debug:   distance(ur5_ee_link) = 1.114483, parent(ur5_ee_link) = ur5_wrist_3_link
Debug:   distance(top_chassis_link) = 0.543339, parent(top_chassis_link) = base_link
Debug:   distance(ur5_upper_arm_link) = 0.089640, parent(ur5_upper_arm_link) = ur5_shoulder_link
Debug:   distance(ur5_wrist_3_link) = 1.099483, parent(ur5_wrist_3_link) = ur5_wrist_2_link
Debug:   distance(hand_eye_camera_link) = 1.156131, parent(hand_eye_camera_link) = ur5_tool0
Debug:   distance(top_plate_rear_link) = 0.716219, parent(top_plate_rear_link) = top_plate_link
Debug:   distance(rear_right_wheel) = 0.928133, parent(rear_right_wheel) = base_link
Debug:   distance(realsense_gripper_left_ir_frame) = 1.177637, parent(realsense_gripper_left_ir_frame) = realsense_gripper_depth_frame
Debug:   distance(front_bumper_link) = 1.031889, parent(front_bumper_link) = base_link
Debug:   distance(realsense_gripper_depth_optical_frame) = 1.177637, parent(realsense_gripper_depth_optical_frame) = realsense_gripper_depth_frame
Debug:   distance(top_plate_front_link) = 0.716219, parent(top_plate_front_link) = top_plate_link
Debug:   distance(ur5_wrist_1_link) = 0.922214, parent(ur5_wrist_1_link) = ur5_forearm_link
Debug:   distance(realsense_gripper_link) = 1.177637, parent(realsense_gripper_link) = realsense_gripper_bottom_screw_frame
Debug:   distance(front_left_wheel) = 0.928133, parent(front_left_wheel) = base_link
Debug:   distance(ur5_tool0) = 1.099483, parent(ur5_tool0) = ur5_wrist_3_link
Debug:   distance(realsense_gripper_right_ir_optical_frame) = 1.227637, parent(realsense_gripper_right_ir_optical_frame) = realsense_gripper_right_ir_frame
Debug:   distance(ur5_base) = 0.300000, parent(ur5_base) = ur5_base_link
Debug:   distance(front_right_wheel) = 0.928133, parent(front_right_wheel) = base_link
Debug:   distance(rear_bumper_link) = 1.031889, parent(rear_bumper_link) = base_link
Debug:   distance(sensor_arch_mount_link) = 0.905744, parent(sensor_arch_mount_link) = top_plate_link
Debug:   distance(gripper_finger1_knuckle_link) = 1.177339, parent(gripper_finger1_knuckle_link) = gripper_base_link
Debug:   distance(base_link) = 0.543339, parent(base_link) = top_plate_link
Debug:   distance(ur5_shoulder_link) = 0.089564, parent(ur5_shoulder_link) = ur5_base_link
Debug:   distance(realsense_gripper_color_optical_frame) = 1.192637, parent(realsense_gripper_color_optical_frame) = realsense_gripper_color_frame
Debug:   distance(gripper_base_link) = 1.114483, parent(gripper_base_link) = ur5_ee_link
Debug:   distance(gripper_finger2_finger_link) = 1.209089, parent(gripper_finger2_finger_link) = gripper_finger2_knuckle_link
Debug:   distance(bumper_link) = 0.543339, parent(bumper_link) = base_link
Debug:   distance(gripper_finger1_finger_link) = 1.209089, parent(gripper_finger1_finger_link) = gripper_finger1_knuckle_link
Debug:   distance(base_footprint) = 0.660619, parent(base_footprint) = base_link
Debug:   distance(gripper_finger1_inner_knuckle_link) = 1.177202, parent(gripper_finger1_inner_knuckle_link) = gripper_base_link
Debug:   distance(gripper_finger2_inner_knuckle_link) = 1.177202, parent(gripper_finger2_inner_knuckle_link) = gripper_base_link
Debug:   distance(realsense_gripper_color_frame) = 1.192637, parent(realsense_gripper_color_frame) = realsense_gripper_depth_frame
Debug:   distance(realsense_gripper_right_ir_frame) = 1.227637, parent(realsense_gripper_right_ir_frame) = realsense_gripper_depth_frame
Debug:   distance(realsense_gripper_left_ir_optical_frame) = 1.177637, parent(realsense_gripper_left_ir_optical_frame) = realsense_gripper_left_ir_frame
Debug:   distance(ur5_base_link) = 0.000000, parent(ur5_base_link) = ur5_base_link
Debug:   distance(ur5_forearm_link) = 0.514912, parent(ur5_forearm_link) = ur5_upper_arm_link
[... cut ...]
cost values: (5.64995e-29, 2.54817e-26, 0, 0, 0, 0.00376516, 0.250176, 2.09428, 1.40089, 0.78796, 0, 0, 0.64266)
constraint violations: (5.64995e-29, 2.54817e-26, 0, 0, 0, 0.00376516, 0.250176, 2.09428, 1.40089, 0.78796, 0, 0)
n func evals: 20
n qp solves: 21
==================
Info:    TrajOptMotionPlannerTask motion planning failed (Failed to find valid solution: OPT_PENALTY_ITERATION_LIMIT) for process input: Tesseract Composite Instruction
Debug:   Error
Info:    Planning took 5.508835 seconds.
`

closed 2026-01-18 by rjoomen