tesseract-robotics/trajopt #420: crash when running the examples in tesseract_ros_examples as shown in the README file

MiaoDragon, 2024-08-09:
Hi,

Thanks for creating the software. I'm a PhD student working on trajectory optimization for robotic manipulation, and I'm trying to test this solver. I have successfully installed everything into a catkin workspace, and have successfully built it. However, I'm facing crashes when running the example code as shown in the README.
For all of the examples, I'm having the error message shown below:
```
Program: }
[ERROR] [1723244402.148066798]: Hit enter key to continue!
rviz: /usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:113: Eigen::internal::variable_if_dynamic<T, Value>::variable_if_dynamic(T) [with T = long int; int Value = 4]: Assertion `v == T(Value)' failed.
[tesseract_ros_examples_puzzle_piece_example_rviz-4] process has died [pid 167334, exit code -6, cmd /opt/ros/noetic/lib/rviz/rviz -d /home/yinglong/Documents/research/task_motion_planning/non-prehensile-manipulation/trajopt_ws/src/tesseract_ros/tesseract_ros_examples/config/examples.rviz __name:=tesseract_ros_examples_puzzle_piece_example_rviz __log:=/home/yinglong/.ros/log/1b12624e-56a3-11ef-93cd-dde98a0e9465/tesseract_ros_examples_puzzle_piece_example_rviz-4.log].
log file: /home/yinglong/.ros/log/1b12624e-56a3-11ef-93cd-dde98a0e9465/tesseract_ros_examples_puzzle_piece_example_rviz-4*.log
```
After I turned off rviz, this error message disappeared, but it still crashed with the following message:
```
auto-starting new master
process[master]: started with pid [170271]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bd11e600-56a3-11ef-93cd-dde98a0e9465
process[rosout-1]: started with pid [170297]
started core service [/rosout]
process[tesseract_ros_examples_pick_and_place_example_node-2]: started with pid [170304]
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
[ERROR] [1723244674.636392258]: Hit enter key to continue!

Program: Composite Instruction, Description: Tesseract Composite Instruction
Program: {
Program:   Move Instruction, Move Type: 1, State WP: Pos=    0     0     0 -1.57     0     0     0
, Description: Start Instruction
Program:   Move Instruction, Move Type: 1, Cart WP: xyz=0.15, 0.4, 1.127
, Description: From start to pick Approach
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.15, 0.4, 0.977
, Description: Pick Approach
Program: }
Info:    Pick plan
[tesseract_ros_examples_pick_and_place_example_node-2] process has died [pid 170304, exit code -11, cmd /home/yinglong/Documents/research/task_motion_planning/non-prehensile-manipulation/trajopt_ws/devel/lib/tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node __name:=tesseract_ros_examples_pick_and_place_example_node __log:=/home/yinglong/.ros/log/bd11e600-56a3-11ef-93cd-dde98a0e9465/tesseract_ros_examples_pick_and_place_example_node-2.log].
log file: /home/yinglong/.ros/log/bd11e600-56a3-11ef-93cd-dde98a0e9465/tesseract_ros_examples_pick_and_place_example_node-2*.log
```
Below are the messages by screenshot.
I'm using the latest version in the master branch for each of the packages. You can find my catkin workspace packages in the screenshot below. Could you help me on this issue and suggest me how to debug it? Thanks and looking forward to your reply!

[image]
[image]
[image]

Levi-Armstrong, 2024-08-10:
Thank you for reporting, I will test and see if I can reproduce.

Levi-Armstrong, 2024-08-11:
The example ran for me without any issues. What version are you current running?

Levi-Armstrong, 2024-08-12:
What version of the Tessseract/Trajopt software are you running?

zzzbbs, 2024-08-12:
> What version of the Tessseract/Trajopt software are you running?

I am running on Trajopt 0.23.2 with Eigen 3.3.7.

zzzbbs, 2024-08-12:
> What version of the Tessseract/Trajopt software are you running?

Other required packages are below TrajOpt's version, around 0.23.1.

MiaoDragon, 2024-08-12:
> The example ran for me without any issues. What version are you current running?

Hi @Levi-Armstrong  I'm using the latest master branch, so 0.23.2. Other tesseract packages are also the latest master branch as for now.

zzzbbs, 2024-08-13:
Hi @Levi-Armstrong @MiaoDragon, I have solved this problem by updating my Eigen package from 3.3.7 to 3.3.9 and run all the provided examples.

Levi-Armstrong, 2024-08-13:
That is interesting, because I am running 3.3.7. I am not sure if you have time, but it might be worth building in debug and running the example with gdb to see exactly what the issue was.

zzzbbs, 2024-08-13:
> That is interesting, because I am running 3.3.7. I am not sure if you have time, but it might be worth building in debug and running the example with gdb to see exactly what the issue was.

I‘m currently busy on my work, but I will try to figure out what causes it exactly during my free time. I also tried Eigen 3.3.8, which did not work either. I guess, it's probably because of the way I installed Eigen 3.3.7 using `sudo apt-get install libeigen3-dev` that caused this.

MiaoDragon, 2024-08-13:
> Hi @Levi-Armstrong @MiaoDragon, I have solved this problem by updating my Eigen package from 3.3.7 to 3.3.9 and run all the provided examples.

@zzzbbs Hi. How did you update Eigen package? Did you update it by installing through apt-get, or did you install it through source? Thanks!

zzzbbs, 2024-08-14:
> > Hi @Levi-Armstrong @MiaoDragon, I have solved this problem by updating my Eigen package from 3.3.7 to 3.3.9 and run all the provided examples.
> 
> @zzzbbs Hi. How did you update Eigen package? Did you update it by installing through apt-get, or did you install it through source? Thanks!

I built it from source using the following command.
`git clone -b 3.3.9 https://gitlab.com/libeigen/eigen.git
mkdir build
cd build/
cmake .. -DCMAKE_BUILD_TYPE=Release
sudo make install
`

MiaoDragon, 2024-08-21:
Hi Thanks for @zzzbbs 's suggestion, now I'm able to run the code without error.

closed 2024-08-21 by MiaoDragon